NVIDIA-ISAAC-ROS/isaac_ros_common #2: Error executing the script

FPSychotic, 2021-11-02:
hi, it is the first time I use docker, so no idea what can be the problem
Jetson Xavier NX
Ubuntu 20.04 with asked me something of a docker daemon and I chosen disable it , but no 100% sure.
Jetpack 4.6

`imother@imother:~/isaac_ros_common-main/scripts$ ./run_dev.sh
isaac_ros_dev not specified, assuming /home/imother/isaac_ros_common-main/scripts
~/isaac_ros_common-main/scripts ~/isaac_ros_common-main/scripts
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
errors pretty printing info
Default docker runtime is not nvidia!, please make sure the following line is
present in /etc/docker/daemon.json
"default-runtime": "nvidia",

And then restart the docker daemon
~/isaac_ros_common-main/scripts
imother@imother:~/isaac_ros_common-main/scripts$ sudo systemctl --now enable docker
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
imother@imother:~/isaac_ros_common-main/scripts$ sudo systemctl restart docker.service
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
imother@imother:~/isaac_ros_common-main/scripts$ sudo systemctl restart docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
imother@imother:~/isaac_ros_common-main/scripts$ sudo systemctl daemon-reload && sudo systemctl restart docker
Job for docker.service failed because the control process exited with error code.
See "systemctl status docker.service" and "journalctl -xe" for details.
imother@imother:~/isaac_ros_common-main/scripts$ ./run_dev.sh
isaac_ros_dev not specified, assuming /home/imother/isaac_ros_common-main/scripts
~/isaac_ros_common-main/scripts ~/isaac_ros_common-main/scripts
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
errors pretty printing info
Default docker runtime is not nvidia!, please make sure the following line is
present in /etc/docker/daemon.json
"default-runtime": "nvidia",
I tried some commands as you see. I did try the script with sudo,same issue, and my docker user is made

And then restart the docker daemon
~/isaac_ros_common-main/scripts
imother@imother:~/isaac_ros_common-main/scripts$ 

`

hemalshahNV, 2021-11-02:
Are you running JetPack 4.6 on your Xavier NX? You mentioned Ubuntu 20.04 but JP 4.6 is based on Ubuntu 18.04. In order to use the Docker container through the run_dev.sh script, you'll need to have setup the [Nvidia Container Toolkit](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html) which will also have you install Docker (the daemon and the client).

FPSychotic, 2021-11-03:
> Are you running JetPack 4.6 on your Xavier NX? You mentioned Ubuntu 20.04 but JP 4.6 is based on Ubuntu 18.04. In order to use the Docker container through the run_dev.sh script, you'll need to have [...]

yes that's right I'm in 20.04 focal with Xavier NX.

fixed with this, your .json text is wrong 

`{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia"
}
`

FPSychotic, 2021-11-03:
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia"
}

closed 2021-11-03 by FPSychotic