NVIDIA-ISAAC-ROS/isaac_ros_common #3: colcon build error using dockerfile

rob-pointcloud, 2021-11-04:
Hi Nvidia team, first a huge thanks for creating and sharing these packages in ROS, I'm very excited to get this working and start development 😄 

I am using an Xavier NX.

- I flashed the SD card with the latest Jetpack 4.6 downloaded online.
- I followed the directions [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_dnn_inference#setup)
     - I created a "workspaces/isaac_ros_common" directory, and git cloned the two required packages
     - I ran "isaac_ros_common/scripts/run_dev.sh your_ws", which completed succesfully

However, when I tried to build for the first time:

`cd /workspaces/isaac_ros-dev
colcon build && . install/setup.bash`

 I received the following error:

`
admin@XavierNX:/workspaces/isaac_ros-dev$ colcon build && . install/setup.bash
Starting >>> isaac_ros_nvengine_interfaces
Starting >>> isaac_ros_test
Starting >>> isaac_ros_common                                                                    
Finished <<< isaac_ros_test [3.20s]                                                                                              
Finished <<< isaac_ros_common [23.8s]                                                                                        
Finished <<< isaac_ros_nvengine_interfaces [31.7s]                          
Starting >>> isaac_ros_nvengine
Starting >>> isaac_ros_dnn_encoders                
--- stderr: isaac_ros_nvengine                                                                                       
/usr/bin/ld:/workspaces/isaac_ros-dev/src/isaac_ros_common/isaac_ros_nvengine/gxf/lib/gxf_jetpack46/core/libgxf_core.so: file format not recognized; treating as linker script
/usr/bin/ld:/workspaces/isaac_ros-dev/src/isaac_ros_common/isaac_ros_nvengine/gxf/lib/gxf_jetpack46/core/libgxf_core.so:1: syntax error
collect2: error: ld returned 1 exit status
make[2]: *** [libgxe_node.so] Error 1
make[1]: *** [CMakeFiles/gxe_node.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< isaac_ros_nvengine [21.0s, exited with code 2]
Aborted  <<< isaac_ros_dnn_encoders [57.5s]                                        

Summary: 3 packages finished [1min 30s]
  1 package failed: isaac_ros_nvengine
  1 package aborted: isaac_ros_dnn_encoders
  1 package had stderr output: isaac_ros_nvengine
  5 packages not processed
admin@XavierNX:/workspaces/isaac_ros-dev$ 
`
Any help would be much appreciated!

hemalshahNV, 2021-11-04:
Thanks for making use of the work! We’re excited to see what you do with it.

Did you happen to install git-lfs first before cloning the repository? In the troubleshooting section of the README, you should find the above error. There are binaries we host in the git repository through git-lfs. The error is because those files have not been downloaded and only the placeholder text file is present.

Running “git lfs pull” should get git-lfs to pull down the files.

rob-pointcloud, 2021-11-04:
:man_facepalming: doh! sorry about that, not sure how I missed it! I had installed git lfs on my Xavier NX, but I had not run the "git lfs pull" in the docker container before the build, that fixed the issue!

closed 2021-11-04 by rob-pointcloud