NVIDIA-ISAAC-ROS/isaac_ros_common #4: Best practice while using docker

kaganGH, 2021-11-07:
Hello,

I would like to ask you a naive practical question about the usage of docker "isaac_ros-dev". After we build the docker and run the container, the container starts running from "/workspaces/isaac_ros-dev/". If we would like to use different ros2 isaac packages such as "isaac_ros_common" and "isaas_ros_argus_camera", should we clone and build these packages in this directory? Basically, as I can see, user "admin" does not have root permissions in other directories and this is adjusted from "run_dev.sh". Is this done on purpose? Thank you in advance.

hemalshahNV, 2021-11-08:
The best practice would be to have one ROS workspace in your home directory that has all of your packages in a src/ directory (e.g., "~/workspaces/myawesome_ros_ws/src/"). When you invoke run_dev.sh, you can provide the directory to mount into the Docker container as "/workspaces/isaac_ros-dev" (e.g., "scripts/run_dev.sh ~/workspaces/myawesome_ros_ws" will be mapped to /workspaces/isaac_ros-dev inside the container). The "admin" user is a non-root user with the same UID as your host user so that you can write files from inside the container or outside the container as the same user.

kaganGH, 2021-11-13:
Ok thank you.

closed 2021-11-13 by kaganGH