NVIDIA-ISAAC-ROS/isaac_ros_common #5: Docker error running run_dev.sh

griz1112, 2021-11-12:
Trying to build the dev env and when I run run_dev.sh I get this error message.
failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown
~/gem_ws/src/isaac_ros_common
nvidia-docker is there nvidia-runtime is enabled. this is on a jetson xavier.

hemalshahNV, 2021-11-12:
The error seems to indicate an issue with the way Docker itself is setup or a mismatch with the nvidia-runtime perhaps. Could you check that you are running JetPack 4.6 by running the following: `cat /etc/nv_tegra_release` which should yield the following? 
> # R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021

griz1112, 2021-11-12:
Thats what it is. I'm on 5.2 will have to reflash thanks.

griz1112, 2021-11-12:
I reflashed it has the proper sdk now and the error is still there.

hemalshahNV, 2021-11-12:
You're right. This looks like an instance of related [issue 1](https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-962597353) and [issue 2](https://forums.developer.nvidia.com/t/docker-containers-wont-run-after-recent-apt-get-upgrade/194369) with the NVIDIA Container Runtime and Docker version mismatch. 

You can either downgrade Docker, use an experimental version of NVIDIA Container Runtime, or add flags  `--security-opt  seccomp=unconfined` to `docker run` command in `run_dev.sh`. We're evaluating the last option as a clean workaround in the meantime.

griz1112, 2021-11-12:
I was seeing something similar when I googled the error but the versions were higher than what is on the Xavier so I figured I was good since they were saying to downgrade to the version that is on the Xavier. I'll try again in a bit and let you know thanks.

griz1112, 2021-11-13:
Is this where I need to add the flag?
# Run container from image
print_info "Running $CONTAINER_NAME"
docker run -it --rm --security-opt seccomp=unconfined  \
    --privileged --network host \
    ${DOCKER_ARGS[@]} \

Getting the same error when I do it this way.

msrasheed, 2021-12-03:
I had a similar issue and downgrading docker.io to 20.10.7-0ubuntu1~18.04.2 fixed the issue. 
Got it to work after following the advice from this thread: https://bugs.launchpad.net/ubuntu/+source/docker.io/+bug/1950071

swapnesh-wani-nvidia, 2021-12-07:
This has been fixed in the [1.7.0 version of nvidia-container-toolkit](https://github.com/NVIDIA/nvidia-container-toolkit/releases/tag/v1.7.0). Instruction on installing the nvidia-container-toolkit can be found [here](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#installation-guide).

hemalshahNV, 2022-01-04:
@swapnesh-wani-nvidia That worked for me!

closed 2022-05-23 by hemalshahNV