NVIDIA-ISAAC-ROS/isaac_ros_common #7: Build errors with galactic

benbarron, 2021-11-23:
Hi, I am trying to use this repo and https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline to do hardware acceleration on a Ros2 (galactic) based system that uses realsense cameras. I am using an AGX with ubuntu 18.04 and jetpack 4.6.1. I am running into install errors when building isaac_ros_common inside my base docker image. I noticed the readme says it is designed to be compatible with ros2 foxy. are there any plans to support ros2 galactic? thanks.

paulhendricks, 2021-11-29:
@hemalshahNV any thoughts on this?

hemalshahNV, 2021-11-29:
We have not tried to build any of the Isaac ROS packages with Galactic, but I would think the changes required are not too bad if you wanted to try. We support LTS Foxy and we're working with OSRF on improvements for Humble. We do not plan on supporting Galactic in the interim unfortunately.

hemalshahNV, 2022-05-23:
We will be supporting ROS2 Humble in our upcoming Isaac ROS release.

closed 2022-05-23 by hemalshahNV