NVIDIA-ISAAC-ROS/isaac_ros_common #11: Having problems pulling the images from the repository

omer-arad, 2022-02-08:
Hi, 
I'm getting this error when trying to run `run_dev.sh` script:

```docker: Error response from daemon: pull access denied for isaac_ros_dev-aarch64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.```

Any idea how to fix this?

hemalshahNV, 2022-02-09:
Was there anything else further up in the log? Usually that last line means the image was not built earlier on.

omer-arad, 2022-02-09:
`~/workspace/isaac_ros_common ~/workspace/isaac_ros_common
Building aarch64 base as image: isaac_ros_dev-aarch64
Sending build context to Docker daemon  17.87MB
Step 1/39 : ARG BASE_IMAGE="dustynv/ros:foxy-ros-base-l4t-r32.6.1"
Step 2/39 : FROM ${BASE_IMAGE}
 ---> bc3b88966255
Step 3/39 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> e36aa92e75f7
Step 4/39 : RUN apt-get update && apt-get install -y     build-essential     cmake     curl     git     lsb-release     sudo     tar     unzip     vim     wget     software-properties-common && rm -rf /var/lib/apt/lists/*
 ---> Running in a587a29e351d
failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown
Running isaac_ros_dev-aarch64-container
Unable to find image 'isaac_ros_dev-aarch64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-aarch64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.`

I hope it helps

hemalshahNV, 2022-02-10:
We have seen the seccomp filter rule issue before. Please take a look at https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/issues/5#issuecomment-987554412 and above to resolve.

omer-arad, 2022-02-10:
I wasn't able to downgrade the version, I re-flashed Jetpack 4.6. 
This time I did not use `sudo apt-get upgrade` at all, and it worked.
I guess there is an issue with one of the recent updates in Ubuntu.

Thank you for the help!

closed 2022-02-10 by omer-arad