NVIDIA-ISAAC-ROS/isaac_ros_common #14: Docker cannot fectch archives

FPSychotic, 2022-02-25:
Sorry but I have not idea what is the problem , because I hate docker and I have no idea how it works. This confirm I have good reasons ejejej. Ubuntu 20.04, driver 510, gtx3060 laptop, cuda 11.6 cudnn 8.2xxx, last VPi, opencv 4.5.2, ros Foxy. I download by 4G LTE.
I tried install previously without docker, just like I did with my NX, but there is 4 packages that wont compile. But that is other history I think.

`Sending build context to Docker daemon  17.87MB
Step 1/36 : ARG BASE_IMAGE=nvcr.io/nvidia/tritonserver:21.08-py3
Step 2/36 : FROM ${BASE_IMAGE}
 ---> 3eb9530b33c0
Step 3/36 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> d39053ad2f2c
Step 4/36 : RUN apt-get update && apt-get install -y         bash-completion         build-essential         clang-format         cmake         curl         git         gnupg2         locales         lsb-release         rsync         software-properties-common         wget         vim         unzip         mlocate && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> f938cf0dfc6d
Step 5/36 : RUN apt-get update && apt-get install -y         python3-flake8         python3-opencv         python3-pip         python3-pytest-cov         python3-setuptools && rm -rf /var/lib/apt/lists/*
 ---> Running in 33c0cf6b3495
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:5 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1580 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Ign:7 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease
Get:8 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:9 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:10 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release [696 B]
Get:11 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [982 kB]
Get:12 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [25.8 kB]
Get:13 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [842 kB]
Get:14 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Release.gpg [836 B]
Get:15 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Packages [611 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [29.4 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1136 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1052 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2008 kB]
Get:21 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [50.8 kB]
Get:22 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [25.2 kB]
Fetched 21.8 MB in 3s (8072 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
python3-setuptools is already the newest version (45.2.0-1).
python3-setuptools set to manually installed.
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
The following additional packages will be installed:
  adwaita-icon-theme at-spi2-core autoconf automake autotools-dev cpp-8 file
  fontconfig fontconfig-config fonts-dejavu-core gcc-8 gcc-8-base gdal-data
  gfortran gfortran-8 gfortran-9 gstreamer1.0-plugins-base
  gtk-update-icon-cache hicolor-icon-theme humanity-icon-theme i965-va-driver
  intel-media-va-driver javascript-common libaacs0 libaec0 libaom0
  libarmadillo9 libarpack2 libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data
  libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3
  libavcodec58 libavformat58 libavutil56 libbdplus0 libblas3 libbluray2
  libcaf-openmpi-3 libcairo-gobject2 libcairo2 libcdparanoia0 libcfitsio8
  libcharls2 libchromaprint1 libcoarrays-dev libcoarrays-openmpi-dev
  libcodec2-0.9 libcolord2 libcups2 libdap25 libdapclient6v5 libdatrie1
  libdc1394-22 libdrm-amdgpu1 libdrm-common libdrm-intel1 libdrm-nouveau2
  libdrm-radeon1 libdrm2 libepoxy0 libepsilon1 libevent-2.1-7
  libevent-core-2.1-7 libevent-dev libevent-extra-2.1-7 libevent-openssl-2.1-7
  libevent-pthreads-2.1-7 libfabric1 libfontconfig1 libfreetype6 libfreexl1
  libfribidi0 libfyba0 libgcc-8-dev libgdal26 libgdcm3.0 libgdk-pixbuf2.0-0
  libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgeos-3.8.0 libgeos-c1v5
  libgeotiff5 libgfortran-8-dev libgfortran-9-dev libgfortran5 libgif7 libgl1
  libgl1-mesa-dri libgl2ps1.4 libglapi-mesa libglvnd0 libglx-mesa0 libglx0
  libgme0 libgraphite2-3 libgsm1 libgstreamer-plugins-base1.0-0 libgtk-3-0
  libgtk-3-bin libgtk-3-common libharfbuzz0b libhdf4-0-alt libhdf5-103
  libhdf5-openmpi-103 libhwloc-dev libhwloc-plugins libhwloc15 libice6
  libigdgmm11 libilmbase24 libjbig0 libjpeg-turbo8 libjpeg8 libjs-jquery
  libjs-jquery-hotkeys libjs-jquery-isonscreen libjs-jquery-metadata
[... 463 lines cut ...]
Get:318 http://archive.ubuntu.com/ubuntu focal/universe amd64 libhwloc-dev amd64 2.1.0+dfsg-4 [205 kB]
Get:319 http://archive.ubuntu.com/ubuntu focal/universe amd64 libopenmpi-dev amd64 4.0.3-0ubuntu1 [798 kB]
Get:320 http://archive.ubuntu.com/ubuntu focal/universe amd64 proj-bin amd64 6.3.1-1 [88.8 kB]
Fetched 194 MB in 27s (7266 kB/s)
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/z/zvbi/libzvbi0_0.2.35-17_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/libb/libbdplus/libbdplus0_0.1.2-3_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/p/pmix/libpmix2_3.1.5-1_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/l/lcms2/liblcms2-2_2.9-4_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libe/libepoxy/libepoxy0_1.5.4-1_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/p/proj/libproj15_6.3.1-1_amd64.deb  501  Not Implemented [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libr/librest/librest-0.7-0_0.8.1-1_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libx/libxrandr/libxrandr2_1.5.2-0ubuntu1_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/j/jquery-throttle-debounce/libjs-jquery-throttle-debounce_1.1+dfsg.1-1_all.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/o/opencv/libopencv-highgui4.2_4.2.0+dfsg-5_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/o/opencv/libopencv-videostab4.2_4.2.0+dfsg-5_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/m/mesa/mesa-va-drivers_21.2.6-0ubuntu0.1~20.04.1_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/n/numpy/python3-numpy_1.17.4-5ubuntu3_amd64.deb  400  Bad Request [IP: 91.189.88.152 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c apt-get update && apt-get install -y         python3-flake8         python3-opencv         python3-pip         python3-pytest-cov         python3-setuptools && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
Running isaac_ros_dev-x86_64-container
Unable to find image 'isaac_ros_dev-x86_64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-x86_64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
~/ros2_isaac_ws/src/isaac_ros_common
igcs@igcs-Legion-5-15ACH6H:~/ros2_isaac`

hemalshahNV, 2022-02-25:
This seems to be an issue pulling down an apt-get cache update, not necessarily a Docker issue. It appears there is trouble fetching from a number of mirrors which prevented the Docker image from getting built. Maybe try a different internet connection I suppose unless those mirrors have gone bad?

`http://archive.ubuntu.com/ubuntu focal/universe amd64 libhwloc-plugins amd64 2.1.0+dfsg-4 [14.4 kB] Err:151 http://archive.ubuntu.com/ubuntu focal/universe amd64 libpmix2 amd64 3.1.5-1 400 Bad Request [IP: 91.189.88.152`

FPSychotic, 2022-02-25:
> This seems to be an issue pulling down an apt-get cache update, not necessarily a Docker issue. It appears there is trouble fetching from a number of mirrors which prevented the Docker image from [...]

Probably is due to I use 4G LTE and something with filtering, DNS , no idea. I have not other conexion available, no idea how to fix it. better way , and my favorite would be compiling , but will fail in four packages. I will post the problem in the right packages

Cheers.

closed 2022-02-25 by FPSychotic