NVIDIA-ISAAC-ROS/isaac_ros_common #15: is there a way to take a snapshot of container and use it later on after running run_dev.sh

altineller, 2022-02-28:
Hello,

./scripts/run_dev.sh takes a long time to prepare the container for use, so what is the proper way of taking a snapshot of running container and using it later on?

in docker there are commits and checkoints. which one would you prefer?

hemalshahNV, 2022-05-23:
When you launch a new container from an image, it needs to perform a series of setup tasks that can only be done in a running state (rosdep install or other). Could you tell us a bit more about what exactly is taking so long just to make sure we're thinking of the same thing?

bblumberg, 2022-07-13:
I actually think that run_dev.sh is pretty quick, but I do think there is a potential issue associated with unintentionally pulling in new code that could break things. Anyway,  is this much more than doing a docker container commit? 

bb

hemalshahNV, 2022-07-13:
rosdep install could yield different outcomes when run each time, that's true. Using the new facilities to change the sequence of docker images built [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common#configuring-run_devsh), you could add running the steps in the entrypoint during docker image build using another Dockerfile rather than docker container launch. Note, however, there are many files that need to be mounted into the container that may not be accessible during docker image build time which will cause problems for you.

bblumberg, 2022-07-14:
Let me give you an example. Suppose I build a docker image
humble.realsense.mine via the run_dev script. Once it has been built, I
understand that when you launch the docker container, you need to pull in
the various -v files and directories, etc. but I still don't get why you
can't pull in the previously built humble.realsense.mine docker image if
all you want to do is do local development within that container. Note,
this is purely to do development within the container rather than build yet
another docker image using it as a base or layer.

Maybe there is another way of achieving the outcome, but from a user
perspective it is important to be able to launch a container without
worrying about it being rebuilt.

bb

hemalshahNV, 2022-07-14:
When run_dev.sh executes, you’re right that it asks Docker to build the image. Docker, however, knows that the build context is unchanged and uses its cached layers, not rebuilding anything even if the result of any of the commands in each Dockerfile would have returned a different result than the first time. 

After the container is “rebuilt,” the container itself runs setup scripts to be available for development which is where rosdep install happens.

bblumberg, 2022-07-14:
Ok, thanks for the clarification. That makes sense.

bb