NVIDIA-ISAAC-ROS/isaac_ros_common #25: Error running run_dev.sh possible version dependency error caused by protobuf

altineller, 2022-06-03:
Hello,

After upgrading to latest jetpack with ubuntu 20.04, running run_dev.sh ~/workspace causes the following error:

> Collecting protobuf>=3.12.2 (from onnx)
  Downloading https://files.pythonhosted.org/packages/6c/be/4e32d02bf08b8f76bf6e59
f2a531690c1e4264530404501f3489ca975d9a/protobuf-4.21.0-py2.py3-none-any.whl (164kB
)
protobuf requires Python '>=3.7' but the running Python is 3.6.9
The command '/bin/bash -c wget https://nvidia.box.com/shared/static/p57jwntv436lfr
d78inwl7iml6p13fzh.whl -O torch-1.8.0-cp36-cp36m-linux_aarch64.whl &&         apt-
get update && apt-get install -y libopenblas-base libopenmpi-dev &&         python
3 -m pip install -U numpy torch-1.8.0-cp36-cp36m-linux_aarch64.whl onnx' returned 
a non-zero code: 1
Running isaac_ros_dev-aarch64-container
Unable to find image 'isaac_ros_dev-aarch64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-aarch64, 
repository does not exist or may require 'docker login': denied: requested access 
to the resource is denied.
See 'docker run --help'.
~/isaac_ros-dev/src/isaac_ros_common/scripts

protobuf complains about the python version, and then the docker image does not get build.

any ideas/recomendations/help on this problem greatly appreciated.

best regards,
can

hemalshahNV, 2022-06-03:
Our upcoming release of Isaac ROS Common will support Jetpack 5 DP (based on Ubuntu 20.04). The currently available version will likely run into a series of versioning mismatches as it was intended for Jetpack 4.6.1 (based on Ubuntu 18.04).

altineller, 2022-06-04:
Thank you very much for your hard work.

When can we expect the upcoming release?

Also, is there any way to run isaac ros common on a regular PC with nvidia graphics card, running ubuntu 20.04, or that would also require ubuntu 18.04

hemalshahNV, 2022-06-04:
Thank you for making good use of it! We're working hard to get the release ready by end of June 2022. Yes, you can use run_dev.sh on Ubuntu 20.04 on an x86_64 system with a discrete NVIDIA GPU.

altineller, 2022-06-05:
@hemalshahNV 

I have tried running run_dev.sh on a jetson agx, installed with Jetpack 4,6,1, and 4.6.2 - and still got the same error:

> protobuf requires Python '>=3.7' but the running Python is 3.6.9
The command '/bin/bash -c wget https://nvidia.box.com/shared/static/p57jwntv436lfrd78inwl7iml6p13fzh.whl -O torch-1.8.0-cp36-cp36m-linux_aarch64.whl &&         apt-get update && apt-get install -y libopenblas-base libopenmpi-dev &&         python3 -m pip install -U numpy torch-1.8.0-cp36-cp36m-linux_aarch64.whl onnx' returned a non-zero code: 1
Running isaac_ros_dev-aarch64-container
Unable to find image 'isaac_ros_dev-aarch64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-aarch64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
~/isaac_ros-dev/src/isaac_ros_common

In my jetson agx /etc/apt/preferences, I have the following statements that disables updating of docker, otherwise you get another error, but this was resolved in the forums.

Package: docker.io
Pin: version 20.10.2*
Pin-Priority: 1001

Package: containerd
Pin: version 1.5.2*
Pin-Priority: 1001

It appears to me it might be a issue with torch. On the other hand, it runs on x86_64 pc. Any ideas, how to get this going on an agx?

Best Regards.
Can

hemalshahNV, 2022-06-06:
We haven't seen this here but you could comment out that line in "docker/Dockerfile.aarch64.base" and proceed without Torch which will cause problems if you're using isaac_ros_triton.

millertheripper, 2022-06-08:
I had the same issue and I was able to solve it by changing the following lines in the docker file: 

**On line 74, add the command to upgrade PIP**
RUN python3 -m pip install --upgrade pip

**Then, find the following line and set the numpy version to 1.19.4, since the latest numpy version will cause problems in later steps during docker build process**
# Install Pytorch and ONNX
RUN wget https://nvidia.box.com/shared/static/p57jwntv436lfrd78inwl7iml6p13fzh.whl -O torch-1.8.0-cp36-cp36m-linux_aarch64.whl && \
        apt-get update && apt-get install -y libopenblas-base libopenmpi-dev && \
        **python3 -m pip install numpy==1.19.4 && \**
        python3 -m pip install -U torch-1.8.0-cp36-cp36m-linux_aarch64.whl onnx

Regards

hemalshahNV, 2022-08-01:
The latest Isaac ROS Common run_dev.sh uses pre-built Docker images so you don't have to build all of this yourself anymore. The PyTorch included is not CUDA-enabled (using public version) which will be fixed in an upcoming release.

closed 2022-09-19 by hemalshahNV