NVIDIA-ISAAC-ROS/isaac_ros_common #27: Question: What are the docker images for?

maxpolzin, 2022-06-29:
Hi,
Thank you for your great work on ROS2 nodes for the Jetson boards.
Reading the aarch64 and amd64 docker image files in the repository, they do not seem to compile either of the isaac_ros_* packages.
What are those docker images used for? Should I use them as a base image and build isaac_ros_* packages on top of them?

maxpolzin, 2022-07-01:
I just noticed the larg update on the repository with commit https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/commit/9da5dc1bef4543d36e8c84aeeb0e60f4baf9329f.

Do you plan to release Docker image with the Isaac ROS packages already built?

hemalshahNV, 2022-07-01:
Yes, we are preparing to release prebuilt containers shortly. We are reviewing the images to make sure all of the open source licensing are being properly respected.

maxpolzin, 2022-07-13:
Are the prebuild containers already released? Where will the release be announced?

hemalshahNV, 2022-07-13:
I'll reply to this thread when ready and there will be a note on the `isaac_ros_common` README.md as well.

hemalshahNV, 2022-07-25:
The prebuilt images have been released and will get automatically used if you update to the latest `isaac_ros_common` release. `run_dev.sh` will pull them down and use them transparently for you. Here's the [link](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/containers/ros/tags) to the Docker images on NGC directly for reference.

maxpolzin, 2022-07-26:
Great. Thx for letting me know. :) 
I will check them out as soon as we need to do apriltag detection again.

closed 2022-08-01 by hemalshahNV