NVIDIA-ISAAC-ROS/isaac_ros_common #30: torchvision compatibility issue...

bblumberg, 2022-07-08:
I am running into a showstopper using this framework on the agx orin. When I used the default docker file, the version of pytorch was incompatible with the latest version of CUDA (whatever is on the latest 5.0.1 jetpack. I subsequently followed the instructions for downloading pytorch from the jetson download center, and that seemed to resolve the CUDA incompatibility, BUT, the version of torchvision that is included in that package gives an error when loaded:
1. open python3
2. import torch
3. import torchvision

and you get:
/usr/local/lib/python3.8/dist-packages/torchvision/io/image.py:13: UserWarning: Failed to load image Python extension: 
  warn(f"Failed to load image Python extension: {e}")

And yolov5 fails when it tries to use a torchvision image operator.

Here are the installed python packages
torch
torch-1.13.0a0+340c4120.nv22.06.dist-info
torchaudio
torchaudio-0.12.0.dist-info
torchgen
torchvision
torchvision-0.13.0.dist-info

This issue does appear on various forums, but the fact that it doesn't exist on the nvidia l4t-pytorch container suggests to me that it is some system dependency that is provided by that container and not by the l4t-base container.

Any help that you can provide would be greatly appreciated, and if you need additional information, please let me know.

Thanks,

bb

hemalshahNV, 2022-07-15:
Thanks for working this out! We'll explore updating our Isaac ROS Common images to build torchvision from source directly.

jaiveersinghNV, 2022-10-03:
The CUDA-enabled version of torch is available in the Isaac ROS Dev base images released in `v0.11.0`, but torchvision was not rebuilt from source.

closed 2022-10-24 by jaiveersinghNV