NVIDIA-ISAAC-ROS/isaac_ros_common #33: Question: Run on Jetson Nano

maxpolzin, 2022-07-13:
Why are the Jetson Xavier NX or Nano not supported? 
Simply because you haven't tried or do they have different architectures?

hemalshahNV, 2022-07-13:
We haven't qualified our packages on these platforms but there is nothing preventing you from running on them. They have the same architecture and OS but with different compute capabilities (memory, CPU/GPU clock speeds). DNN models that you could load on an AGX Xavier, for example, may not load successfully on a Nano.

maxpolzin, 2022-07-13:
Great thx for your fast response. If I have time, I try to run some nodes on a Nano and let you know how it goes.

closed 2022-07-13 by maxpolzin

jiefengsun, 2023-09-23:
@maxpolzin Have you tried? Does it work on Jetson Nano?

maxpolzin, 2023-09-24:
Hi @jiefengsun, nope, I haven't had time to try it out. Please let me know how it goes if you do.