NVIDIA-ISAAC-ROS/isaac_ros_common #38: Docker args need to include the '/usr/local/cuda-10.2' libs for v0.9.1ea2

dirksavage88, 2022-08-13:
This is an issue only when building isaac_ros_image_proc: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline/issues/18

Docker args need to include the cuda libraries in the run.dev.sh script:

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/b5261b2d7cd3a1e9dfb857ad5a0664766dbd385c/scripts/run_dev.sh#L134

append "-v /usr/local/cuda-10.2/targets/aarch64-linux/lib/:/usr/local/cuda-10.2/targets/aarch64-linux/lib/"

For reference: https://issueantenna.com/repo/nvidia-isaac-ros/isaac_ros_argus_camera/issues/5

@hemalshahNV Are these early access repo's getting any support for disconnected issues across isaac_ros_* packages dependent on each other? Seems like this probably was due to building the packages up to isaac_ros_common but not building isaac_ros_image_pipeline alongside of it in the workspace. Either that or various script discrepancies across multiple DP/EA versions...

hemalshahNV, 2022-08-15:
Could you confirm what Jetpack version you're running ? With JP 4.6 installed on host (JP 4.6.1 may not be compatible with EA2), you should not have needed to manually mount host CUDA libraries into the container. They should be available through LD_LIBRARY_PATH: [Dockerfile.aarch64.base](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/b5261b2d7cd3a1e9dfb857ad5a0664766dbd385c/docker/Dockerfile.aarch64.base#L116) for EA2 

```
ENV LD_LIBRARY_PATH="/usr/local/cuda-10.2/targets/aarch64-linux/lib:${LD_LIBRARY_PATH}"
```
Are the CUDA files mounted at the above path inside the container?

Could you try again with the latest Developer Preview release or, if moving to Jetpack 5.x is not a possibility, could you try with EA3 instead (the last ROS2 Foxy on JP 4.6.1)? We are not able to support any previous Early Access releases of Isaac ROS.

dirksavage88, 2022-08-20:
I have that path added to my host, yet it doesn't transfer to the container even with the explicit environment path set. Ownership/permissions issue perhaps?

closed 2022-08-20 by dirksavage88