NVIDIA-ISAAC-ROS/isaac_ros_common #39: error building the docker container branch: release-dp

Aki1608, 2022-08-20:
Hi, 

after running `./run_dev.sh ~/isaac_ws`, it shows this error. 

`E: Unable to locate package tensorrt    
E: Unable to locate package vpi2-dev`

Why is it not finding these two packages? I tried installing several times but same result.

exact error:

```RUN apt-get update && apt-get install -y     tensorrt     vpi2-dev && rm -rf /var/lib/apt/lists/* && apt-get clean
 ---> Running in f9da819c424b
Get:1 https://repo.download.nvidia.com/jetson/common r34.1 InRelease [2519 B]
Get:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal/restricted arm64 Packages [1317 B]
Get:4 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease [24.4 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 Packages [1234 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports focal/universe arm64 Packages [11.1 MB]
Get:10 http://ports.ubuntu.com/ubuntu-ports focal/multiverse arm64 Packages [139 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports focal-updates/main arm64 Packages [1802 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe arm64 Packages [1116 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports focal-updates/restricted arm64 Packages [4571 B]
Get:14 http://ports.ubuntu.com/ubuntu-ports focal-updates/multiverse arm64 Packages [9067 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports focal-backports/main arm64 Packages [54.8 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports focal-backports/universe arm64 Packages [27.5 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 Packages [823 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted arm64 Packages [4330 B]
Get:19 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 Packages [1419 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports focal-security/multiverse arm64 Packages [3253 B]
Fetched 18.4 MB in 6s (3189 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package tensorrt
E: Unable to locate package vpi2-dev
The command '/bin/bash -c apt-get update && apt-get install -y     tensorrt     vpi2-dev && rm -rf /var/lib/apt/lists/* && apt-get clean' returned a non-zero code: 100
Failed to build base image: isaac_ros_dev-aarch64, aborting.

swapnesh-wani-nvidia, 2022-08-22:
Please try the [latest](https://github.com/NVIDIA-ISAAC-ROS/.github/blob/main/profile/release-notes.md#isaac-ros-0101-dp-developer-preview-july-12-2022) Isaac ROS DP. `run_dev.sh` will pull a pre-built docker image and you don't have to build one yourself.

Aki1608, 2022-08-27:
Yes. It solved the issue. Thank you. I am closing this.

closed 2022-08-27 by Aki1608