NVIDIA-ISAAC-ROS/isaac_ros_common #40: Docker on release jetpack 5.0.2

automech-rb, 2022-08-22:
Hi,

When using prebuild docker image provided in this repo on jp5.0.2, building various packages (for eg. urg_node) leads to error shown below:
```
Failed to find exported target names in '/opt/ros/humble/install/share/urg_node_msgs/cmake/export_urg_node_msgs__rosidl_generator_cExport.cmake'
```

Also, as pointed in other issue, building image from start leads to following error:
```
E: Unable to locate package tensorrt
E: Unable to locate package vpi2-dev
```
Do you know any way to solve above?
Thanks

swapnesh-wani-nvidia, 2022-08-22:
Please have a look at this #39 

Look for the existence of the `Dockerfile.aarch64.humble.nav2` which references the pre-built image.

automech-rb, 2022-08-23:
Hi,

I am able to use  pre-built image but it leads to build error on many packages which were building successfully on jp 5.0.1
Here is one of the example:
use CONFIG_IMAGE_KEY=humble.nav2.realsense.urg
and content of urg image as:
```
ARG BASE_IMAGE
FROM ${BASE_IMAGE}

# Install URG node
RUN apt-get update && mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
    && git clone -b master https://github.com/ros-drivers/urg_node_msgs.git \
    && git clone -b ros2-devel https://github.com/ros-perception/laser_proc.git \
    && git clone -b ros2-devel https://github.com/ros-drivers/urg_c.git \
    && git clone -b ros2-devel https://github.com/ros-drivers/urg_node.git \
    && source ${ROS_ROOT}/setup.bash && cd ${ROS_ROOT} \
    && rosdep install -y -r --ignore-src --from-paths src --rosdistro ${ROS_DISTRO} \
    && colcon build --merge-install --parallel-workers 6 --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-up-to urg_node \
    && rm -Rf src logs build \
&& rm -rf /var/lib/apt/lists/* \
&& apt-get clean
```

This leads to error
```
--- stderr: urg_node
CMake Error at /opt/ros/humble/install/share/urg_node_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/opt/ros/humble/install/share/urg_node_msgs/cmake/export_urg_node_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /opt/ros/humble/install/share/urg_node_msgs/cmake/urg_node_msgsConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)
```
And similar **Failed to find exported target names in** errors on other packages!

automech-rb, 2022-08-23:
Found the issue. It is related to to [this ](https://github.com/dusty-nv/jetson-containers/issues/181)issue on dusty jetson containers.

hemalshahNV, 2022-08-23:
Yes, we have run into this and had reverted to cmake 3.23 to get builds running again. That fix will be released with new updated prebuilt images in our upcoming release this month (August).

hemalshahNV, 2022-09-02:
This should be resolved in our latest release where cmake is downgraded to 3.23.2.

closed 2022-09-02 by hemalshahNV