NVIDIA-ISAAC-ROS/isaac_ros_common #41: Clarification on Dockers

javadan, 2022-08-25:
Hi,

I'm a bit confused by the instructions.

I've cloned the repo and `git-lfs`'ed, so there is `~/your_ws/src/isaac_ros_common` now.

If I go to `~/your_ws/src/isaac_ros_common`, and run `./scripts/run_dev.sh ~/your_ws`
it builds `isaac_ros_dev-aarch64` with base `aarch64-humble-nav2-image`

I see there's `isaac_ros_common/docker/realsense-dockerfile-example/` with `Dockerfile.realsense`, and the two jetsonHacks scripts.

From what I understand of the instructions, if I wanted to build realsense on top, 
I would put these lines in `~/your_ws/src/isaac_ros_common/scripts/.isaac_ros_common-config`:

CONFIG_IMAGE_KEY=humble.nav2.realsense
CONFIG_DOCKER_SEARCH_DIRS=($HOME/src/isaac_ros_common/docker/)

Then from `~/your_ws/src/isaac_ros_common`, I run `./scripts/run_dev.sh ~/your_ws`
I tried the external directory name, too, in case it's meant to be that.  Tried with the realsense-dockerfile-example directory too.

But it just builds using key `aarch64.humble.nav2.user`, and gives no indication that the config had any effect.  

Could someone please clarify the usage? Thanks

hemalshahNV, 2022-09-01:
All of the steps as you described them are spot on. `.isaac_ros_common-config` should be a sibling to `run_dev.sh` and set the `CONFIG_IMAGE_KEY` to `humble.nav2.realsense` which all looks correct. `CONFIG_DOCKER_DEARCH_DIRS=($HOME/src/isaac_ros_common/docker/realsense-dockerfile-example)` should let it resolve Dockerfile.realsense and the scripts within, but still doesn't explain why `run_dev.sh` is not picking up your config file. Could you list the exact directory structure you have here?

javadan, 2022-09-29:
Thanks, my Jetson is inside a robot now, which a production house is going to hold on to, so unfortunately I won't be able to test this out.  But hopefully I'll grab an Orin Nano next year, and give it another shot.  Will close for now.  Thanks for the help.

closed 2022-09-29 by javadan