NVIDIA-ISAAC-ROS/isaac_ros_common #45: Example of isaac_ros packages layers on top of base image to setup the environment

arainbilal, 2022-11-10:
Hi, I need some guidance in adding layers (isaac_ros packages) on top of the base images provided in Isaac ROS to setup the environment. As an example, I have used the attached `Dockerfile.nvblox` to be available in the image. 
[Dockerfile.nvblox.txt](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/files/9977794/Dockerfile.nvblox.txt)

I have followed the following steps, as mentioned in the README:
1. Added the `CONFIG_IMAGE_KEY="humble.nav2.nvblox"`
2. run_dev.sh

The config image key was resolved as ` x86_64.humble.nav2.nvblox.user` which is expected. I think that there are no root permissions to create a folder is `/opt`, therefore, the image failed and shows the following error: 

```
Cloning into 'isaac_ros_nvblox'...
Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
My question is more about the best practices to create images that already have isaac_ros packages required for the development. I would appreciate it if you may include an example of a ROS package, in particular, that requires the CUDA architectures such as Nvblox, etc. Thanks in advance.

swapnesh-wani-nvidia, 2022-11-15:
Looks like the url for cloning nvblox is not setup for https protocol. You can try to use this url in you docker file
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox

arainbilal, 2022-11-15:
Hi @swapnesh-wani-nvidia 
Thanks for your help. I have changed the URL in my Dockerfile. The cloning of the nvblox passed through. However, the CMAKE_CUDA_ARCHITECTURES is empty. I can confirm that this is not the case when manually cloning the repository in the container. Just wondering if you can guide me furthermore on this. Thanks.

```
Starting >>> nvblox
Starting >>> nvblox_msgs
Finished <<< nvblox_msgs [5.88s]
--- stderr: nvblox
CMake Warning at thirdparty/stdgpu/stdgpu.cmake:49 (message):
  Falling back to default CCs :
Call Stack (most recent call first):
  CMakeLists.txt:55 (include)

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_cuda_check".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_cuda_check".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_gpu_hash".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_lib".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_lib".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in examples/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "esdf_query".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in examples/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "esdf_query".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in examples/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "esdf_query".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in tests/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_test_utils".
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in tests/CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "nvblox_test_utils".
This warning is for project developers.  Use -Wno-dev to suppress it.

---
[Processing: nvblox]
[Processing: nvblox]
Finished <<< nvblox [1min 27s]
Starting >>> nvblox_ros
[Processing: nvblox_ros]
Finished <<< nvblox_ros [37.4s]

Summary: 3 packages finished [2min 5s]
  1 package had stderr output: nvblox
```

hemalshahNV, 2022-11-22:
Can you report the output of `cmake --version` inside your container? It should read `cmake version 3.23.2`

arainbilal, 2022-11-25:
Hi @hemalshahNV, Thank you for your response. The cmake version is  `3.23.2`. To reproduce this problem, please use the following
1.
```
mkdir tmp && cd tmp
git clone --recursive https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common.git
cd isaac_ros_common && git lfs pull
``` 
2. The `.isaac_ros_common-config` include `CONFIG_IMAGE_KEY="humble.nav2.nvblox"`. 
3. Add the attached docker file, and rename the file, as follows:
```
cd docker
mv Dockerfile.nvblox.txt Dockerfile.nvblox
```
4. Run the script
```
cd ../scripts && ./run_dev.sh
```
5. List the docker images, which should have the following:
```
isaac_ros_dev-x86_64
nvblox-image
```
Please let me know if I an provide more information to reproduce the above warnings. Thanks.

hemalshahNV, 2023-01-31:
@arainbilal , are you perhaps running this on a host with no GPU or without configuring the default runtime as `nvidia-container-toolkit` for docker which makes the GPUs are made available to the Docker build environment to resolve the current GPU architecture to build the CUDA code for? The build files for nvblox only compile the CUDA code for the architecture for the host GPU to save compilation time.

closed 2023-02-28 by jaiveersinghNV