NVIDIA-ISAAC-ROS/isaac_ros_common #47: run_dev.sh fails with `failed to register layer: Error processing tar file(exit status 1): archive/tar: invalid tar header`

haowu80s, 2022-12-16:
```
Building /home/hao-wu/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.humble.nav2 as image: x86_64-humble-nav2-image with base: 
Sending build context to Docker daemon   80.9kB
Step 1/1 : FROM nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b
x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b: Pulling from nvidia/isaac/ros
3b65ec22a9e9: Pulling fs layer 
fd80d866e8b2: Pulling fs layer 
a364ca75fd6d: Pulling fs layer 
3d4731d03623: Waiting 
53a5c2e0251f: Waiting 
b00ff40d02d9: Waiting 
3036e9b94123: Waiting 
453fdcdda788: Waiting 
35e12ec5e515: Waiting 
11f61a475a23: Waiting 
24280cf31c9a: Waiting 
79007799e2ed: Waiting 
03eb76abf1e5: Waiting 
4f4fb700ef54: Waiting 
8c80ae8980cd: Waiting 
3918bec26dcf: Waiting 
5988980892dd: Waiting 
f4e902364789: Pulling fs layer 
9e630d0e3180: Waiting 
140ee16889c1: Waiting 
da88e5d79c6f: Waiting 
f61455589e43: Waiting 
4363df7b85e8: Pulling fs layer 
6e5e8533bd39: Pulling fs layer 
f61455589e43: Download complete 
0dc9580c2401: Download complete 
15be8ac18616: Download complete 
2d20876aec99: Download complete 
b27a26b8bc54: Download complete 
77e8e1895580: Download complete 
58486e24d700: Download complete 
ef9047d558d3: Download complete 
2020f0da6e65: Download complete 
e89c028f35a6: Download complete 
57dc833364e9: Download complete 
0e8234224d83: Download complete 
ae4316a64573: Download complete 
460a2092d34a: Download complete 
25577f5f0c34: Download complete 
470a4f1a3861: Download complete 
c70aab8b800b: Download complete 
b2c35c86bea9: Download complete 
b2acdf4d640f: Download complete 
627909d01b7c: Download complete 
0a410ef4d0a2: Download complete 
1e68f15edef6: Download complete 
ab3a937b2ce7: Download complete 
b707c805a446: Download complete 
799b55167afb: Download complete 
f85fee33b522: Download complete 
4eac5839291b: Download complete 
69b60e038821: Download complete 
5376045668ac: Download complete 
edff3e3858c0: Download complete 
be17fae500ce: Download complete 
b2b1d900c1aa: Download complete 
82ebd5cb8096: Download complete 
349cd3bf8b52: Download complete 
failed to register layer: Error processing tar file(exit status 1): archive/tar: invalid tar header
```

haowu80s, 2022-12-16:
```
docker version
Client: Docker Engine - Community
 Version:           20.10.21
 API version:       1.41
 Go version:        go1.18.7
 Git commit:        baeda1f
 Built:             Tue Oct 25 18:01:58 2022
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.21
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.18.7
  Git commit:       3056208
  Built:            Tue Oct 25 17:59:49 2022
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.6.10
  GitCommit:        770bd0108c32f3fb5c73ae1264f7e503fe7b2661
 nvidia:
  Version:          1.1.4
  GitCommit:        v1.1.4-0-g5fd4c4d
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

haowu80s, 2022-12-16:
```
docker info
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Docker Buildx (Docker Inc., v0.9.1-docker)
  compose: Docker Compose (Docker Inc., v2.12.2)
  scan: Docker Scan (Docker Inc., v0.21.0)

Server:
 Containers: 4
  Running: 0
  Paused: 0
  Stopped: 4
 Images: 30
 Server Version: 20.10.21
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc io.containerd.runc.v2 io.containerd.runtime.v1.linux nvidia
 Default Runtime: nvidia
 Init Binary: docker-init
 containerd version: 770bd0108c32f3fb5c73ae1264f7e503fe7b2661
 runc version: v1.1.4-0-g5fd4c4d
 init version: de40ad0
 Security Options:
  apparmor
  seccomp
   Profile: default
  cgroupns
 Kernel Version: 5.15.0-56-generic
 Operating System: Ubuntu 22.04.1 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 24
 Total Memory: 31.24GiB
 Name: haowu-System-Product-Name
 ID: UDGT:MUPJ:5YYV:6UD6:3MEA:OSZF:T74W:BBT7:RQ6B:HBC4:HES2:HVD4
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```

hemalshahNV, 2022-12-16:
Very strange. It seems that docker wasn't able to pull the image from the nvcr.io Docker registry. This could have been a transient failure though. Could you try running `docker pull nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b` and see if that succeeds (basically what the script was doing here which failed)?

haowu80s, 2022-12-16:
I tried it in a different environment and it works. Must be some issue on my side. I'm closing the issue. Thanks!

closed 2022-12-16 by haowu80s