NVIDIA-ISAAC-ROS/isaac_ros_common #49: pip install --no-cache-dir, instead of --no-cache?

iraadit, 2023-01-11:
In https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.aarch64#L149, `pip install --no-cache` is used.
`pip install --no-cache` seems to not exist, should it be changed to `pip install --no-cache-dir` ?

hemalshahNV, 2023-01-11:
Strange indeed. `--no-cache-dir` is what we need, but `--no-cache- works fine in `pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)` within the container and when building image itself on a Jetson. Are you rebuilding the aarch64 image or just using the commands? If the latter, could you list what version of pip3 you have (`pip3 --version`)?

iraadit, 2023-01-16:
`# pip3 --version
pip 22.2.2 from /usr/local/lib/python3.8/dist-packages/pip (python 3.8)`

I created a modified version of isaac_ros_common for our needs, and use several recent docker features. It is based on `nvcr.io/nvidia/l4t-base:r35.1.0`, corresponding to isaac_ros_common Isaac ROS 0.11.0 (DP1.1).

I stumbled upon this when I was looking into pip caching.

I modified it to use `docker build bake` and `QEMU`, and it can therefore build on the Jetson, on a host x64 computer and with our CI.

In the [pip documentation](https://pip.pypa.io/en/stable/topics/caching/#disabling-caching), I find "--no-cache-dir", but there is no mention of "--no-cache", no look on Google either.

"--no-cache" can be an argument of `docker build` though

hemalshahNV, 2023-01-31:
We'll update to `--no-cache-dir` in Isaac ROS Common as that reads more correct anyway. We would love to hear about your modifications and add them back into Isaac ROS Common if you'd care to describe them. We're working to enable BuildKit for Isaac ROS Dev base images to gain access to more advanced Docker directives.

executter, 2024-07-19:
pip use argparse to parse arguments. By default, argparse allows long options to be abbreviated. These commands work too:
`pip install --no-cache-d`
`pip install --no-ca`