NVIDIA-ISAAC-ROS/isaac_ros_common #50: GPG key failure on apt update NO_PUBKEY 42D5A192B819C5DA

ikhann, 2023-01-22:
I am trying to run apt update.

```
Sending build context to Docker daemon  43.05MB
Step 1/14 : ARG BASE_IMAGE
Step 2/14 : FROM ${BASE_IMAGE}
 ---> 3671296a1bac
Step 3/14 : RUN apt-get update && apt-get install -y         curl         gnupg2         lsb-release &&         rm -rf /var/lib/apt/lists/* &&         apt-get clean
 ---> Running in 27f60d4ba1cd
Hit:2 http://archive.ubuntu.com/ubuntu focal InRelease
Get:3 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Hit:5 http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu focal InRelease
Get:6 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Hit:7 http://packages.ros.org/ros2-testing/ubuntu focal InRelease
Hit:1 https://repo.download.nvidia.com/jetson/x86_64/focal r35.1 InRelease
Get:8 https://apt.kitware.com/ubuntu focal InRelease [15.5 kB]
Hit:9 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease
Err:8 https://apt.kitware.com/ubuntu focal InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 42D5A192B819C5DA
Hit:10 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease
Reading package lists...
W: GPG error: https://apt.kitware.com/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 42D5A192B819C5DA
E: The repository 'https://apt.kitware.com/ubuntu focal InRelease' is not signed.
The command '/bin/bash -c apt-get update && apt-get install -y         curl         gnupg2         lsb-release &&         rm -rf /var/lib/apt/lists/* &&         apt-get clean' returned a non-zero code: 100
Failed to build base image: isaac_ros_dev-x86_64, aborting.
```

It fails due to some GPG error. Any image I based on this image will have errors running apt update, which means that apt update && ... always fails.

After quick web surfing, I found a popular fix:

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 42D5A192B819C5DA
sudo apt-get update
```

But it didn't help..

ikhann, 2023-01-22:
Solved.
Added these lines before `apt update` in the Dockerfile of my image:

```
RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null \
    && echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | tee /etc/apt/sources.list.d/kitware.list >/dev/null
```

tonynajjar, 2023-01-22:
Thanks that fixed it for me too

hemalshahNV, 2023-01-22:
Thanks, @ikhann! The key from Kitware seems to have been invalidated. We're rebuilding the pre-built Isaac ROS Dev base images with the updated keys. In the meantime, your fix will work if you add it to `Dockerfile.user`  or your custom Dockerfile.

hemalshahNV, 2023-01-30:
We've pushed out updated Isaac ROS Dev base image with the new GPG key. Please try again when you can with the prebuilt image and let us know if you still have any issues.

ikhann, 2023-02-07:
> We've pushed out updated Isaac ROS Dev base image with the new GPG key. Please try again when you can with the prebuilt image and let us know if you still have any issues.
                                           
Hi!
Yes, now everything is perfect. I don't have any errors.
Thank you!

closed 2023-02-07 by ikhann