NVIDIA-ISAAC-ROS/isaac_ros_common #67: Failed to open run_dev.sh in offline mode

tonaaskarwur, 2023-04-27:
I am using NVIDIA Orin and running the run_dev.sh script to open the docker from the main branch (commited 20/04/2023).
 When Orin is connected to the internet, the docker is executed and I can enter the docker. But after the initial pull, the script fails when there is no internet connection.
[With this version](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/tree/4c6309b74f5b2f62eaca203a67a9928b5d8a35ef), I can run the docker without internet connection. 
I couldn't find a line that changes this behavior, as it clearly goes to the different image.
I attached the docker logs here. Thank you
[Docker_log_27.04.txt](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/files/11339662/Docker_log_27.04.txt)

Isopod00, 2023-05-03:
I have this issue as well

hemalshahNV, 2023-05-08:
We switched to the new default build engine for Docker called BuildKit which seems to require an internet connection to validate checksums for image layers. We're following the issue here: https://github.com/docker/buildx/issues/738.

closed 2023-05-09 by tonaaskarwur