NVIDIA-ISAAC-ROS/isaac_ros_common #72: Error while buiding using Docker from run_dev.sh

Avi241, 2023-05-25:
I am getting this following error while running run_dev.sh
`> [1/1] FROM nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_c59eda1b21db0e3c8f28c8d436ad1f4e@sha256:886f1e34b89de38451c8088c8654c95602f185ef65df69cf54459df147234489:
------
failed commit on ref "layer-sha256:d42fd8f99cbae064cb4fd43c1e957ed75af67f344f5c27c091eaddb610ef35bc": "layer-sha256:d42fd8f99cbae064cb4fd43c1e957ed75af67f344f5c27c091eaddb610ef35bc" failed size validation: 4504682496 != 4709419862: failed precondition
Failed to build base image: isaac_ros_dev-x86_64, aborting.`

Can anyone please help regarding this ?

Thanks

jaiveersinghNV, 2023-05-31:
Could you please try running this again on the latest [DP3.1 Release](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/tree/v0.31.0-dp)?

Avi241, 2023-06-07:
I guess it was some network issue I have pulled that same image using docker pull and again run the process it worked fine.

closed 2023-06-08 by Avi241