NVIDIA-ISAAC-ROS/isaac_ros_common #74: Cannot install git-lsf

reinzler, 2023-06-04:
Hi, cannot install git-lfs on Ubuntu 20.04 (Jetson Orin Nano), even after curl -s https://packagecloud.io/install/repositories/github/git-lfs/script.deb.sh | sudo bash. Please, help. Thanks in advance

reinzler, 2023-06-04:
Manage to install :
sudo apt-get install git-lfs
[sudo] password for nvidia: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git-lfs is already the newest version (2.9.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

But now different error: 
nvidia@tegra-ubuntu:~$ git lfs install --skip-repo
/usr/local/bin/git-lfs: 1: ELF: not found
/usr/local/bin/git-lfs: 2: Q�t�e@�: not found
/usr/local/bin/git-lfs: 2: Q: not found
/usr/local/bin/git-lfs: 1: Syntax error: ";" unexpected
nvidia@tegra-ubuntu:~$

reinzler, 2023-06-05:
This helped:
https://askubuntu.com/questions/1434789/running-ubuntu-22-04-jammy-and-failing-to-install-git
	sudo bash -c 'dpkg --configure -a && apt update && apt upgrade'
	sudo apt-get install git-lfs
https://github.com/git-lfs/git-lfs/issues/1063

jaiveersinghNV, 2023-06-05:
Sounds like you were able to resolve your problem. If you have any other issues, please open a ticket in the appropriate repository and we'll be happy to help you.

closed 2023-07-26 by jaiveersinghNV