NVIDIA-ISAAC-ROS/isaac_ros_common #79: Provide OpenCV with CUDA support for the Docker image.

jmrp98, 2023-06-23:
I see that the OpenCV in the Docker image provided does not support CUDA, would it be possible to supply OpenCV with CUDA support? For example I have been using dusty-nv's  ROS2 PyTorch images, which support OpenCV with CUDA, but when I try to build the Isaac Packages with his container, the build fails, therefore I cannot use them and I need to use this ones.  https://github.com/dusty-nv/jetson-containers

jaiveersinghNV, 2023-06-26:
We are working on upgrading to using a CUDA-compatible version of OpenCV, and we'll update this thread when we have it ready.

jmrp98, 2023-09-20:
I was able to modify the aarch64 and ros2_humble Dockerfiles to install a CUDA-compatible version of OpenCV. Should I submit a pull request ?

JiangboJumbo, 2023-10-23:
> I was able to modify the aarch64 and ros2_humble Dockerfiles to install a CUDA-compatible version of OpenCV. Should I submit a pull request ?

Hi! I was also struggling in having CUDA-compatible version of OpenCV inside this container. Could you give some hints on how you managed to do this? I would be so appreciated.

omerts, 2024-01-04:
@jmrp98 Can you please share how you did it?

Raessan, 2024-02-12:
Hello,

Are there any updates about the CUDA-supported openCV in the container? I saw that a new version of CV-CUDA that supports aarch64 has been released (https://github.com/CVCUDA/CV-CUDA/releases/tag/v0.5.0-beta). However, I still wasn't able to create the main image of Isaac ROS using OpenCV with CUDA support.

Thank you in advance.