NVIDIA-ISAAC-ROS/isaac_ros_common #82: 401 unauthorized error while running run_dev.sh

altineller, 2023-07-11:
Hello,

At one point I got the following error:
 => [internal] load build definition from Dockerfile.x86_64.ros2_humble                                                                                                                                     0.0s
 => => transferring dockerfile: 57B                                                                                                                                                                           0.0s
 => [internal] load .dockerignore                                                                                                                                                                             0.0s
 => => transferring context: 2B                                                                                                                                                                               0.0s
 => ERROR [internal] load metadata for nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_1930bbec7e7704243656b695b9df7844                                                                                           0.4s
------
 > [internal] load metadata for nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_1930bbec7e7704243656b695b9df7844:
------
failed to solve with frontend dockerfile.v0: failed to create LLB definition: failed to authorize: rpc error: code = Unknown desc = failed to fetch anonymous token: unexpected status: 401 Unauthorized
Failed to build base image: isaac_ros_dev-x86_64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts

Is this a temporary repository problem or am i requesting an image that does not exist?

Best Regards,
Can

rem821, 2023-07-12:
Can confirm I'm dealing with the exact same problem at the moment. No solution yet.

altineller, 2023-07-12:
@rem821 I found out why:

altineller, 2023-07-12:
I am closing this issue, since it is not an issue anymore.

nakai-omer, 2023-07-13:
@hemalshahNV This is similar to https://forums.developer.nvidia.com/t/build-error-with-isaac-ros-visual-slam/243967/11?u=omers

@altineller You should not need to login in order to build/pull docker images, this is an issue that should be solved. Login is just a workaround.

jaiveersinghNV, 2023-07-24:
We are still working on restoring anonymous Docker image pulls from the NVIDIA Container Registry.

For now, the best workaround is still registering for an NGC account, as mentioned by @altineller above.

brian7989, 2024-02-23:
I registered for NGC account and logged in successfully via `docker login nvcr.io` and tried `ngc config set` but still having same 401 issues. Attaching screenshots for reference.

[image]
[image]

Tuntenfisch, 2025-07-31:
I'm also experiencing this issue sporadically. Sometimes it works without issue for days. Sometimes I keep getting "401 unauthorized" many times in a row. 

I would really prefer not having to log into nvcr.io because the container is running on various different robots in my university's lab that are also accessible to students.