NVIDIA-ISAAC-ROS/isaac_ros_common #90: BuildKit error on Jetpack 5.1.1 Xavier

jainRAV, 2023-09-06:
When running ./run_dev.sh on newly installed and updated (apt) jetpack, following error occurs: 

```
ERROR: BuildKit is enabled but the buildx component is missing or broken.
       Install the buildx component to build images with BuildKit:
       https://docs.docker.com/go/buildx/
Failed to build base image: isaac_ros_dev-aarch64, aborting.
```

javijurado, 2023-09-06:
I have the same problem, I hope someone can help us

automech-rb, 2023-09-07:
For now you can uninstall docker (Ubuntu version) and install docker-ce from docker website as a temporary solution.

tonaaskarwur, 2023-09-08:
also happened for me with fresh new flashed Orin with Jetpack 5.1
Installing the docker-ce through apt worked fine so far

jaiveersinghNV, 2023-10-23:
Could you please try again with the [latest release](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/issues/90) on JetPack 5.1.2? We believe a simple reinstall should fix the issues with Docker BuildKit

yijh0611, 2023-11-02:
I am experiencing the same issue.
I am using JetPack 5.1.2, Jetson Orin NX.
When I do
``` bash
cd ~/workspaces/isaac_ros-dev/src/isaac_ros_common/ && \
  git lfs ls-files -s
```
I get
``` bash
>   git lfs ls-files -s
945a1a82ee * docker/tao/tao-converter-aarch64-tensorrt8.4.zip (36 KB)
```
This doesn't look like the example on the link. Is this related to the problem?

yijh0611, 2023-11-02:
> For now you can uninstall docker (Ubuntu version) and install docker-ce from docker website as a temporary solution.

This solved the problem. Thank you!

ikhann, 2023-11-06:
> When running ./run_dev.sh on newly installed and updated (apt) jetpack, following error occurs:
>
> ```
> ERROR: BuildKit is enabled but the buildx component is missing or broken.
>        Install the [...]

Assigning `DOCKER_BUILDKIT=0` [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/scripts/build_base_image.sh#L73C1-L73C16) may temporarily solve the issue.

hemalshahNV, 2023-11-06:
Disabling BuildKit will work temporarily but will soon break as we take advantage of BuildKit features + syntax which are not backwards compatible with the legacy builder environment. The issue here is that buildx plugin is no longer installed by default with docker (for some reason). 

As @jaiveersinghNV mentioned, we have a troubleshooting note on this issue [here](https://nvidia-isaac-ros.github.io/troubleshooting/dev_env.html#docker-buildx-plugin-not-installed) that should help here. If not, please let us know.

closed 2023-11-27 by jaiveersinghNV