NVIDIA-ISAAC-ROS/isaac_ros_common #91: ROS docker compilation error

jmaranad, 2023-09-06:
docker image: nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_33836e394da2d095a59afd2d151038f8
device: Jetson Orin NX
            # R35 (release), REVISION: 3.1, GCID: 32827747, BOARD: t186ref, EABI: aarch64, DATE: Sun Mar 19 15:19:21 UTC 2023

Error:
-- stderr: isaac_ros_common                                                                  
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "Findvpi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vpi", but
  CMake did not find one.

  Could not find a package configuration file provided by "vpi" with any of
  the following names:

    vpiConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "vpi" to CMAKE_PREFIX_PATH or set "vpi_DIR"
  to a directory containing one of the above files.  If "vpi" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< isaac_ros_common [3.75s, exited with code 1]

jaiveersinghNV, 2023-10-23:
Could you please try this again on our [latest release](https://nvidia-isaac-ros.github.io/releases/index#isaac-ros-2-0-0-october-18-2023)?

jmaranad, 2023-10-24:
I reinstall my jetson orin jetpack and it solve the problem.

closed 2023-10-24 by jmaranad