NVIDIA-ISAAC-ROS/isaac_ros_common #93: Docker container error.

jiefengsun, 2023-09-25:
It terminated at middle. 
I restarted it. But the following error persists. 

jiefeng@turtlenano:~/workspaces/isaac_ros-dev/src/isaac_ros_common$ ./scripts/run_dev.sh 
isaac_ros_dev not specified, assuming /home/jiefeng/workspaces/isaac_ros-dev
Error: Failed to call git rev-parse --git-dir --show-toplevel: "fatal: not a git repository (or any of the parent directories): .git\n"
Building aarch64.ros2_humble.user base as image: isaac_ros_dev-aarch64 using key aarch64.ros2_humble.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: aarch64.ros2_humble.user
/home/jiefeng/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/jiefeng/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble
Building /home/jiefeng/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble as image: aarch64-ros2_humble-image with base: 
[+] Building 0.7s (3/3) FINISHED                                                                     
 => [internal] load build definition from Dockerfile.aarch64.ros2_humble                        0.1s
 => => transferring dockerfile: 58B                                                             0.0s
 => [internal] load .dockerignore                                                               0.1s
 => => transferring context: 2B                                                                 0.0s
 => ERROR [internal] load metadata for nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_33836e394d  0.3s
------
 > [internal] load metadata for nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_33836e394da2d095a59afd2d151038f8:
------
failed to solve with frontend dockerfile.v0: failed to create LLB definition: rpc error: code = Unknown desc = failed to parse /home/jiefeng/.docker/.token_seed: unexpected end of JSON input
Failed to build base image: isaac_ros_dev-aarch64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common

jiefengsun, 2023-09-25:
I've loged in nvcr.io follwing: https://docs.omniverse.nvidia.com/isaacsim/latest/install_container.html#isaac-sim-requirements-isaac-sim-container

But it still does not work.

swapnesh-wani-nvidia, 2023-10-02:
It looks like your docker is in a weird state. Have a look at this post.

closed 2023-10-30 by jaiveersinghNV