NVIDIA-ISAAC-ROS/isaac_ros_common #95: ros2-rqt no plugins

daodao9801, 2023-10-11:
Hi,

May I know how can I run rqt with standard plugins in this docker? When I finish setting up and run rqt in docker as suggested:
`cd ~/workspaces/isaac_ros-dev/src/isaac_ros_common && \
  ./scripts/run_dev.sh`

`source /workspaces/isaac_ros-dev/install/setup.bash`

`rqt`

There are only 2 tabs in the _Plugins_ ,"Container" and "Visualization". 
The issue is kind of like #88, but no answers are given yet. Thanks in advance.

Pheo

jaiveersinghNV, 2023-10-23:
Could you please update your installation to use the [latest release](https://nvidia-isaac-ros.github.io/releases/index#isaac-ros-2-0-0-october-18-2023)?

With this latest release, we have added an Isaac ROS Buildfarm for building precompiled Debian packages for standard ROS packages. You can install the `rqt_common_plugins` package with the following command:

```
sudo apt-get install ros-humble-rqt-common-plugins
```

daodao9801, 2023-10-26:
Thanks. Sorry I am new to this, may I know if I should run the apt install in the container or not? In another word, should I run the install command after running `./scripts/run_dev.sh` and getting into the docker?

jaiveersinghNV, 2023-10-30:
Yes, you'll want to install the RQT plugins *inside* the container.

closed 2023-12-05 by jaiveersinghNV