NVIDIA-ISAAC-ROS/isaac_ros_common #101: run_dev.sh: using `--gpus` instead of `--runtime nvidia`

Interpause, 2023-11-06:
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/6d3c5c00e0e2b3fc1d75eb4286848d23b05d6dca/scripts/run_dev.sh#L195-L203

I noticed `run_dev.sh`'s Docker container works (`torch.cuda.is_available()` returns True) if I replace `--runtime nvidia` with `--gpus all`. I also noticed in the dev environment setup guide (https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html) that `nvidia-container-runtime` is deprecated. Is using `--gpus all` more suitable on newer versions of Docker?

hemalshahNV, 2023-11-06:
`--gpus all` should enable the same runtime behavior but need to confirm with the nvidia-container-runtime engineers. Thanks for the heads up.

Buddies-as-you-know, 2023-11-17:
Has this been changed?

jaiveersinghNV, 2023-11-20:
@Buddies-as-you-know , could you confirm what version of the CUDA Drivers you have installed? The missing `libnvidia-ml.so.1` library should be included as part of a proper CUDA installation.