NVIDIA-ISAAC-ROS/isaac_ros_common #104: run_dev.sh got two errors can not solved

BaixiangZ, 2023-11-30:
Hi,

Have tried for couple of days with run_dev.sh, constantly got this two error,
first one, 
**"Error: Failed to call git rev-parse --git-dir: exit status 128 "**
I saw in previously closed issues, this should have been fixed, but I still got this problem
Second one,
**Unable to find image 'isaac_ros_dev-x86_64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-x86_64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.**
Btw, this one I modified a little bit in the run_dev.sh, by adding 'sudo' before docker run, as I found that in my case I could only run docker with gpu as sudo (tried on other images/containers). 

I tried different ways but not helped, I am not experienced in docker/Isaac-Ros, very appriciate if  anyone can help.

My system is Ubuntu 20.04 focal.
Best.

growingpenguin, 2024-01-03:
Hi, I am facing the same issue on the first error. Did you receive a solution or guidance regarding this specific problem? Thanks.
My system is Ubuntu 20.04 focal as well. 

```
$ cd ${ISAAC_ROS_WS}/src/isaac_ros_common &&
> ./scripts/run_dev.sh
isaac_ros_dev not specified, assuming /home/rosexpert1/workspaces/isaac_ros-dev
Error: Failed to call git rev-parse --git-dir: exit status 128 
```

jaiveersinghNV, 2024-01-09:
The `git rev-parse` message is an innocuous error message; we use that command to test whether any Git LFS files are present.
 
The [relevant line](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/scripts/run_dev.sh#L94C26-L94C26) in `run_dev.sh` redirects all output from that command to `/dev/null`, so I'm not sure why you're seeing that message output to your terminal:
```
git rev-parse &>/dev/null
```

Are you having any issues actually starting the container, or is this message merely an annoyance?

BaixiangZ, 2024-02-01:
> The `git rev-parse` message is an innocuous error message; we use that command to test whether any Git LFS files are present.
>
> The [relevant [...]

Hi jaiveersingh,

Really thanks for reply, if the 'git rev-parse' is a harmless error message, then the major problem should be that I can not start the docker image, either from the 'run_dev.sh', or just use 'sudo docker run -it isaac_ros_dev-x86_64:latest',

The error message is 'Unable to find image 'isaac_ros_dev-x86_64:latest' locally
docker: Error response from daemon: pull access denied for isaac_ros_dev-x86_64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.'

Hope this information means something, and thanks in advance for help.

Best

BaixiangZ, 2024-03-06:
Hi all, I upgraded from ubuntu 20 to ubuntu 22 and it works now, seems the docer can only run on 22 with humble

growingpenguin, 2024-03-08:
> The `git rev-parse` message is an innocuous error message; we use that command to test whether any Git LFS files are present.
>
> The [relevant [...]

It was merely an annoyance thanks. Works perfectly well for me!