NVIDIA-ISAAC-ROS/isaac_ros_common #106: "File has unexpected size" when running run_dev.sh; Failed to build base image

eddieliao, 2023-12-25:
When running `run_dev.sh`, it fails to build the base image due to a "File has unexpected size" error. I've tried running it at different times, but it has produced the same error over the past two days.

The full log is provided below:
```terminal
Building x86_64.ros2_humble.user base as image: isaac_ros_dev-x86_64 using key x86_64.ros2_humble.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: x86_64.ros2_humble.user
/home/eddie/Documents/GitHub/robomaster_cv/src/isaac_ros/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/eddie/Documents/GitHub/robomaster_cv/src/isaac_ros/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.ros2_humble
Building /home/eddie/Documents/GitHub/robomaster_cv/src/isaac_ros/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.ros2_humble as image: x86_64-ros2_humble-image with base: 
[+] Building 3.4s (5/5) FINISHED                                                                                                                   docker:default
 => [internal] load build definition from Dockerfile.x86_64.ros2_humble                                                                                      0.0s
 => => transferring dockerfile: 567B                                                                                                                         0.0s
 => [internal] load .dockerignore                                                                                                                            0.0s
 => => transferring context: 2B                                                                                                                              0.0s
 => [internal] load metadata for nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_f70fbf3e86d9ae99b527f8cc2c40007b                                                3.4s
 => CACHED [1/1] FROM nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_f70fbf3e86d9ae99b527f8cc2c40007b@sha256:34b51480113ed7383d5bac961c4d3108c10244e4b90b1292b  0.0s
 => exporting to image                                                                                                                                       0.0s
 => => exporting layers                                                                                                                                      0.0s
 => => writing image sha256:9533c30f8f7a44ef934476006d26d6592781b6b78460081198093b66a2859d7a                                                                 0.0s
 => => naming to docker.io/library/x86_64-ros2_humble-image                                                                                                  0.0s
Building /home/eddie/Documents/GitHub/robomaster_cv/src/isaac_ros/isaac_ros_common/scripts/../docker/Dockerfile.user as image: isaac_ros_dev-x86_64 with base: x86_64-ros2_humble-image
[+] Building 5.5s (6/14)                                                                                                                           docker:default
 => [internal] load .dockerignore                                                                                                                            0.0s
 => => transferring context: 2B                                                                                                                              0.0s
 => [internal] load build definition from Dockerfile.user                                                                                                    0.0s
 => => transferring dockerfile: 1.95kB                                                                                                                       0.0s
 => [internal] load metadata for docker.io/library/x86_64-ros2_humble-image:latest                                                                           0.0s
 => [internal] load build context                                                                                                                            0.0s
 => => transferring context: 179B                                                                                                                            0.0s
 => CACHED [ 1/10] FROM docker.io/library/x86_64-ros2_humble-image                                                                                           0.0s
 => ERROR [ 2/10] RUN apt-get update && apt-get install -y         sudo         udev && rm -rf /var/lib/apt/lists/* && apt-get clean                         5.5s
------                                                                                                                                                            
 > [ 2/10] RUN apt-get update && apt-get install -y         sudo         udev && rm -rf /var/lib/apt/lists/* && apt-get clean:                                    
1.031 Get:1 http://packages.ros.org/ros2-testing/ubuntu focal InRelease [4,685 B]                                                                                 
1.205 Get:3 https://isaac.download.nvidia.com/isaac-ros/ubuntu/main focal InRelease [4,316 B]                                                                     
1.205 Get:4 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease [1,581 B]                                                       
1.226 Get:5 http://packages.ros.org/ros2-testing/ubuntu focal/main amd64 Packages [1,191 kB]                                                                      
1.232 Hit:6 http://ppa.launchpad.net/mosquitto-dev/mosquitto-ppa/ubuntu focal InRelease
1.232 Hit:7 http://archive.ubuntu.com/ubuntu focal InRelease
1.232 Get:8 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
1.498 Get:10 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
1.512 Get:11 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Packages [1,317 kB]
1.515 Get:12 https://isaac.download.nvidia.com/isaac-ros/ubuntu/main focal/main amd64 Packages [499 kB]
1.515 Err:12 https://isaac.download.nvidia.com/isaac-ros/ubuntu/main focal/main amd64 Packages
1.515   File has unexpected size (499551 != 499404). Mirror sync in progress? [IP: 93.184.215.166 443]
1.515   Hashes of expected file:
1.515    - Filesize:499404 [weak]
1.515    - SHA256:0bd951d8f3a67be2865c64c2d950577b2158395d8d7fb5ead5e98ea8b9c968f9
1.515    - SHA1:5c3b1c5a796288db5c84c0be663012ad2deeaa11 [weak]
1.515    - MD5Sum:2b6f13316f99af58757494902eb70c78 [weak]
1.515   Release file created at: Thu, 30 Nov 2023 18:17:36 +0000
1.696 Get:13 https://apt.kitware.com/ubuntu focal InRelease [15.5 kB]
1.964 Get:14 https://apt.kitware.com/ubuntu focal/main amd64 Packages [74.7 kB]
2.153 Get:9 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease [28.0 kB]
2.176 Get:15 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
2.192 Get:16 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [3,283 kB]
2.390 Get:2 https://repo.download.nvidia.com/jetson/x86_64/focal r35.4 InRelease [3,302 B]
2.447 Get:17 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [3,761 kB]
2.615 Get:18 https://packagecloud.io/github/git-lfs/ubuntu focal/main amd64 Packages [3,505 B]
3.452 Get:19 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [32.0 kB]
3.454 Get:20 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1,444 kB]
3.548 Get:21 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [3,130 kB]
3.919 Get:22 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [3,279 kB]
4.169 Get:23 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [29.3 kB]
4.170 Get:24 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [1,149 kB]
4.761 Fetched 19.1 MB in 5s (4,196 kB/s)
4.761 Reading package lists...
5.438 E: Failed to fetch https://isaac.download.nvidia.com/isaac-ros/ubuntu/main/dists/focal/main/binary-amd64/Packages.gz  File has unexpected size (499551 != 499404). Mirror sync in progress? [IP: 93.184.215.166 443]
5.438    Hashes of expected file:
5.438     - Filesize:499404 [weak]
5.438     - SHA256:0bd951d8f3a67be2865c64c2d950577b2158395d8d7fb5ead5e98ea8b9c968f9
5.438     - SHA1:5c3b1c5a796288db5c84c0be663012ad2deeaa11 [weak]
5.438     - MD5Sum:2b6f13316f99af58757494902eb70c78 [weak]
5.438    Release file created at: Thu, 30 Nov 2023 18:17:36 +0000
[... cut ...]
  18 | >>> RUN apt-get update && apt-get install -y \
  19 | >>>         sudo \
  20 | >>>         udev \
  21 | >>> && rm -rf /var/lib/apt/lists/* \
  22 | >>> && apt-get clean
  23 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c apt-get update && apt-get install -y         sudo         udev && rm -rf /var/lib/apt/lists/* && apt-get clean" did not complete successfully: exit code: 100
Failed to build base image: isaac_ros_dev-x86_64, aborting.
```
I'm on a laptop running Ubuntu 22.04 if that helps.

ni-r1, 2023-12-28:
Even I am facing the same issue, Could anyone provide a solution or guidance regarding this specific problem? Thanks.

hjc0106, 2023-12-29:
@eddieliao  check your /etc/apt/source.list. Did you add "deb https://isaac.download.nvidia.cn/isaac-ros/ubuntu/main focal main"?

eddieliao, 2023-12-29:
No I haven't; am I supposed to? Why would I need `deb https://isaac.download.nvidia.cn/isaac-ros/ubuntu/main focal main` in my local apt source list if it's supposed to be running in a Docker container? I don't see this mentioned anywhere in the dev environment setup guide either.

SFhmichael, 2023-12-29:
Hi,
I had the same problem with you, although I don’t know why this issue come from and whether it is important or not. I had modified file Dockerfile.user line18 from RUN apt-get update && apt-get install -y \ to RUN apt-get update || true && apt-get install -y \ to avoid error during apt update and the container was created anyway.
Hope this help you and someone can explain why this happened.

SFhmichael, 2023-12-29:
It seems this is a real problem. Although I can build the image and run container, I failed to install some packages by apt install.
$ sudo apt-get install -y ros-humble-isaac-ros-dope ros-humble-isaac-ros-tensor-rt ros-humble-isaac-ros-dnn-image-encoder
Reading package lists… Done
Building dependency tree
Reading state information… Done
E: Unable to locate package ros-humble-isaac-ros-dope
E: Unable to locate package ros-humble-isaac-ros-tensor-rt
E: Unable to locate package ros-humble-isaac-ros-dnn-image-encoder

hjc0106, 2023-12-29:
@SFhmichael I am facing the same issue... I cannot apt install any ros-humble-isaac-* packages due to package location error...

ni-r1, 2023-12-29:
@SFhmichael, after I changed the docker.user file as mentioned above, I am facing the similar  problem with installing the packages!

naslevente, 2025-05-13:
@SFhmichael I had a similar issue installing the ros-humble-isaac-ros-centerpose package. I noticed that starting the container, the `/etc/apt/sources.list` file contains the `deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0` line. With this however, I was getting the same `Unable to locate package` error. Changing that line to `deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0 universe` fixed the issue for me.