NVIDIA-ISAAC-ROS/isaac_ros_common #117: [Question] How to use issac ROS in the yocto image

bchoineubility, 2024-04-19:
Hi,

I am using Yocto system now. so, I hope to know how to use issac ROS in the system built in the yocto.
Is there any guide OR Doc for it? 

BR,
Mark

poett1, 2024-04-25:
Seems like there has not been much effort on this topic in the past, but I'm interested too. 
Suppose you're targeting Jetson hardware and want to build all the nodes using Bitbake. In that case, you can start by adding all the dependencies from [Base](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.aarch64) and [ROS2 Humble](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/Dockerfile.ros2_humble) Dockerfiles on top of [OE4T/tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) / [meta-ros](https://github.com/ros/meta-ros) and create recipes for all the Nvidia Isaac ROS2 nodes (could be quite a lot of work). I could imagine that several parties are interested in this; maybe [meta-tegra-community](https://github.com/OE4T/meta-tegra-community) is a suitable place to discuss this. 

Another way would be to add Docker support to your Yocto image and run the containers provided by Nvidia. However, I don't know if this method supports all Jetson hardware features.

bchoineubility, 2024-04-25:
@poett1 

Many thanks. 

I have built meta-ros with humble version, I can build ros-image-coire but, ros-image-world yet, because there are many not implemented packages for ROS2-humble. so, I have to consider whether I can use issac ros without ros-packaes on the Yocto. 

I got the answer from ros disclosure also I can not use issac ROS docker without ROS humble system on the host(yocto image).

BR,
Mark

closed 2024-04-25 by bchoineubility