NVIDIA-ISAAC-ROS/isaac_ros_common #127: Building error with realsense support

zhanghua7099, 2024-06-11:
Hi! 

I tried to build the docker image with realsense support. I follow the method from this document:

https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html

When I build the docker image:
```
cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
./scripts/run_dev.sh -d ${ISAAC_ROS_WS}
```
It will report the error that it can not find `ros-humble-realsense2-camera`. My computer is x86 platform. 

Do you have some suggestions?

zhanghua7099, 2024-06-11:
The error log is detailed as follows:
```
13.08 W: http://repo.download.nvidia.com/jetson/x86_64/jammy/dists/r36.3/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
13.08 W: https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
13.10 Reading package lists...
13.64 Building dependency tree...
13.75 Reading state information...
13.76 Package ros-humble-realsense2-camera is not available, but is referred to by another package.
13.76 This may mean that the package is missing, has been obsoleted, or
13.76 is only available from another source
13.76 
13.76 E: Version '4.54.1-1jammy.20240517.191635' for 'ros-humble-realsense2-camera' was not found
------
Dockerfile.realsense:20
--------------------
  19 |     # Install realsense2-camera ROS package and librealsense from binaries for x86_64 only
  20 | >>> RUN --mount=type=cache,target=/var/cache/apt \
  21 | >>>     if [[ "$(uname -m)" == "x86_64" ]]; then \
  22 | >>>         mkdir -p /etc/apt/keyrings && \
  23 | >>>         curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | \
  24 | >>>         tee /etc/apt/keyrings/librealsense.pgp > /dev/null  && \
  25 | >>>         echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] \
  26 | >>>         https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
  27 | >>>         tee /etc/apt/sources.list.d/librealsense.list && apt-get update && \
  28 | >>>         apt-get install -y \
  29 | >>>         librealsense2-dkms=${LIBREALSENSE_DKMS_DEB_VERSION} \
  30 | >>>         librealsense2-utils=${LIBREALSENSE_DEB_VERSION} \
  31 | >>>         librealsense2-dev=${LIBREALSENSE_DEB_VERSION} \
  32 | >>>         ros-humble-realsense2-camera=${REALSENSE2_CAMERA_ROS_DEB_VERSION}; \
  33 | >>>     fi
  34 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c if [[ \"$(uname -m)\" == \"x86_64\" ]]; then         mkdir -p /etc/apt/keyrings &&         curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp |         tee /etc/apt/keyrings/librealsense.pgp > /dev/null  &&         echo \"deb [signed-by=/etc/apt/keyrings/librealsense.pgp]         https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main\" |         tee /etc/apt/sources.list.d/librealsense.list && apt-get update &&         apt-get install -y         librealsense2-dkms=${LIBREALSENSE_DKMS_DEB_VERSION}         librealsense2-utils=${LIBREALSENSE_DEB_VERSION}         librealsense2-dev=${LIBREALSENSE_DEB_VERSION}         ros-humble-realsense2-camera=${REALSENSE2_CAMERA_ROS_DEB_VERSION};     fi" did not complete successfully: exit code: 100
~/workspaces/isaac_ros-dev/src/isaac_ros_common
```

zilingzhang, 2024-06-13:
Encountered same issue building x86 isaac container with realsense support. Building on Jetson Orin is fine.

zhanghua7099, 2024-06-14:
Hi! @zilingzhang 

Thanks for your solutions! The docker image can be built. 

However, a new error occurs. When I try to build the ros2 package:
```
colcon build --symlink-install --packages-up-to-regex realsense*
```
It will output the following messages:
```
Starting >>> realsense2_camera_msgs
--- stderr: realsense2_camera_msgs                              
ERROR setuptools_scm._file_finders.git listing git files failed - pretending there aren't any
---
Finished <<< realsense2_camera_msgs [1.19s]
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [0.09s]                                                             
Finished <<< realsense2_camera [0.35s]                     

Summary: 3 packages finished [1.74s]
  1 package had stderr output: realsense2_camera_msgs
```

This problem can be solved by downgrading the version of `setuptools_scm`.
```
pip install setuptools_scm==5.0.0
```
Hope this might be useful.

closed 2024-06-14 by zhanghua7099