NVIDIA-ISAAC-ROS/isaac_ros_common #130: Isaac ROS RealSense Setup: Launch Docker container fail

hvtung26, 2024-06-15:
### Hi, Im trying to build with Realsense set up with Ubuntu x8664 22.04, RTX 4080 and meet error while execute this command:

**cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
./scripts/run_dev.sh -d ${ISAAC_ROS_WS}_**

**Error detail:**

 => ERROR [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/is  3.7s                                                                                                                
 > [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list ||     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list     && apt-get update:                                                                                 
0.229 Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).                            
0.495 OK
0.570 deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0
0.832 Get:1 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy InRelease [3429 B]
0.856 Hit:2 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64  InRelease
1.393 Ign:3 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
1.413 Hit:4 http://archive.ubuntu.com/ubuntu jammy InRelease
1.419 Get:5 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]
1.430 Get:3 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages [29.8 kB]

### **.1430 Err:3 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
1.430   File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 192.229.232.31 443]
1.430   Hashes of expected file:
1.430    - Filesize:29756 [weak]
1.430    - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
1.430    - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
1.430    - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
1.430   Release file created at: Sat, 15 Jun 2024 03:14:19 +0000**
1.691 Get:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
2.881 Hit:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
2.956 Fetched 260 kB in 2s (112 kB/s)
2.956 Reading package lists...
**### 3.640 W: https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
3.640 E: Failed to fetch https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/release-3.0/binary-amd64/Packages.gz  File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 192.229.232.31 443]
3.640    Hashes of expected file:
3.640     - Filesize:29756 [weak]
3.640     - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
3.640     - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
3.640     - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
3.640    Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
3.640 E: Some index files failed to download. They have been ignored, or old ones used instead.**

Dockerfile.x86_64:31
  30 |     # Add Isaac apt repository
  31 | >>> RUN --mount=type=cache,target=/var/cache/apt \
  32 | >>>     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - && \
  33 | >>>     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list || \
  34 | >>>     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list 
  35 | >>>     && apt-get update
  36 |     
**ERROR: failed to solve: process "/bin/bash -c wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" /etc/apt/sources.list ||     echo \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" | tee -a /etc/apt/sources.list     && apt-get update" did not complete successfully: exit code: 100
~/workspaces/isaac_ros-dev/src/isaac_ros_common**

I've noticed there is new version release (3.0.1) an tried to clone ver 3.0 but it does not work. Very appreciated for any help!

HappySamuel, 2024-07-09:
Hi @hvtung26 

Do you have solution for this issue already? I am currently facing the same issue.

Best,
Samuel

hvtung26, 2024-07-09:
Hi @HappySamuel , at that time the error was solved by the technical team after a few days, maybe some mismatch after update new version. Maybe you should create a new issue on issac ros forum