NVIDIA-ISAAC-ROS/isaac_ros_common #131: failed running run_dev.sh

DailyNir, 2024-06-16:
using x86_64 system to install cummotion according to instructions here: https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_cumotion/isaac_ros_cumotion_moveit/index.html#quickstart .
running this line:
cd ${ISAAC_ROS_WS}/src/isaac_ros_common &&    ./scripts/run_dev.sh

gets the following errors:
Launching Isaac ROS Dev container with image key x86_64.ros2_humble.user: /home/nir/workspaces/isaac_ros-dev/
Building x86_64.ros2_humble.user base as image: isaac_ros_dev-x86_64
Building layered image for key x86_64.ros2_humble.user as isaac_ros_dev-x86_64
Using configured docker search paths: /home/nir/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker
Checking if base image nvcr.io/isaac/ros:x86_64-ros2_humble-user_ec7307b12dd30efb6f3f0633da81be95 exists on remote registry
Checking if base image nvcr.io/isaac/ros:x86_64-ros2_humble_ca64dc26066b1bcb639e16978c53ff75 exists on remote registry
Checking if base image nvcr.io/isaac/ros:x86_64_5ac6becb55810a5b2ed3ea0d61801133 exists on remote registry
Resolved the following 3 Dockerfiles for target image: x86_64.ros2_humble.user
/home/nir/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/nir/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble
/home/nir/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64
Building /home/nir/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64 as image: x86_64-image with base: 
[+] Building 4.4s (7/19)                                         docker:default
 => [internal] load build definition from Dockerfile.x86_64                0.0s
 => => transferring dockerfile: 5.97kB                                     0.0s
 => [internal] load metadata for nvcr.io/nvidia/tritonserver:23.10-py3     1.8s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [stage-0  1/16] FROM nvcr.io/nvidia/tritonserver:23.10-py3@sha256:835  0.0s
 => CACHED [stage-0  2/16] RUN mkdir -p /opt/nvidia/isaac_ros_dev_base &&  0.0s
 => CACHED [stage-0  3/16] RUN --mount=type=cache,target=/var/cache/apt a  0.0s
 => ERROR [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     2.6s
------                                                                          
 > [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list ||     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list     && apt-get update:
0.161 Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
0.613 OK
0.665 deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0
0.955 Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease
0.999 Hit:2 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64  InRelease
1.010 Hit:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
1.023 Hit:4 http://security.ubuntu.com/ubuntu jammy-security InRelease
1.066 Get:5 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy InRelease [3429 B]
1.078 Hit:6 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
1.864 Ign:7 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
1.936 Get:7 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages [29.8 kB]
1.936 Err:7 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
1.936   File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 93.184.221.222 443]
1.936   Hashes of expected file:
1.936    - Filesize:29756 [weak]
1.936    - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
1.936    - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
1.936    - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
1.936   Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
1.979 Fetched 3429 B in 1s (2769 B/s)
1.979 Reading package lists...
2.609 W: https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
2.609 E: Failed to fetch https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/release-3.0/binary-amd64/Packages.gz  File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 93.184.221.222 443]
2.609    Hashes of expected file:
2.609     - Filesize:29756 [weak]
2.609     - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
2.609     - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
2.609     - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
2.609    Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
2.609 E: Some index files failed to download. They have been ignored, or old ones used instead.
------
Dockerfile.x86_64:31
--------------------
  30 |     # Add Isaac apt repository
  31 | >>> RUN --mount=type=cache,target=/var/cache/apt \
  32 | >>>     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - && \
  33 | >>>     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list || \
  34 | >>>     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list \
  35 | >>>     && apt-get update
  36 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" /etc/apt/sources.list ||     echo \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" | tee -a /etc/apt/sources.list     && apt-get update" did not complete successfully: exit code: 100
~/workspaces/isaac_ros-dev/src/isaac_ros_common

liuyt, 2024-06-16:
i also meet the error.

 => [stage-0  2/16] RUN mkdir -p /opt/nvidia/isaac_ros_dev_base && dpkg-query -W | sort > /opt/nvidia/isaac_ros_dev_base/x86_64-start-packages.csv                                        45.8s
 => [stage-0  3/16] RUN --mount=type=cache,target=/var/cache/apt apt-get update && apt-get install -y         software-properties-common && add-apt-repository universe && apt-get updat  19.1s
 => ERROR [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF "deb https:  39.7s
------
 > [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list ||     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list     && apt-get update:
0.219 Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
1.176 OK
1.255 deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0
1.988 Hit:1 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64  InRelease
2.069 Hit:2 http://security.ubuntu.com/ubuntu jammy-security InRelease
2.069 Get:3 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy InRelease [3429 B]
3.190 Ign:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
3.294 Get:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages [29.8 kB]
3.294 Err:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
3.294   File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 198.18.0.53 443]
3.294   Hashes of expected file:
3.294    - Filesize:29756 [weak]
3.294    - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
3.294    - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
3.294    - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
3.294   Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
31.61 Ign:5 http://archive.ubuntu.com/ubuntu jammy InRelease
31.61 Ign:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
31.61 Ign:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
32.61 Ign:5 http://archive.ubuntu.com/ubuntu jammy InRelease
32.61 Ign:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
32.61 Ign:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
34.61 Ign:5 http://archive.ubuntu.com/ubuntu jammy InRelease
34.61 Ign:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
34.61 Ign:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
38.61 Err:5 http://archive.ubuntu.com/ubuntu jammy InRelease
38.61   Could not connect to archive.ubuntu.com:80 (198.18.0.9), connection timed out
38.61 Err:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
38.61   Unable to connect to archive.ubuntu.com:80:
38.61 Err:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
38.61   Unable to connect to archive.ubuntu.com:80:
38.64 Fetched 3429 B in 37s (91 B/s)
38.64 Reading package lists...
39.60 W: https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
39.60 W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/jammy/InRelease  Could not connect to archive.ubuntu.com:80 (198.18.0.9), connection timed out
39.60 W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/jammy-updates/InRelease  Unable to connect to archive.ubuntu.com:80:
39.60 W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/jammy-backports/InRelease  Unable to connect to archive.ubuntu.com:80:
39.60 E: Failed to fetch https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/release-3.0/binary-amd64/Packages.gz  File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 198.18.0.53 443]
39.60    Hashes of expected file:
39.60     - Filesize:29756 [weak]
39.60     - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
39.60     - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
39.60     - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
39.60    Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
39.60 W: Some index files failed to download. They have been ignored, or old ones used instead.
------
Dockerfile.x86_64:31
--------------------
  30 |     # Add Isaac apt repository
  31 | >>> RUN --mount=type=cache,target=/var/cache/apt \
  32 | >>>     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - && \
  33 | >>>     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list || \
  34 | >>>     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list \
  35 | >>>     && apt-get update
  36 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" /etc/apt/sources.list ||     echo \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" | tee -a /etc/apt/sources.list     && apt-get update" did not complete successfully: exit code: 100

ashutoshramola, 2024-06-17:
Same problem in x86-64 system:

`~/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts$ ./run_dev.sh $ISAAC_ROS_WS 
Launching Isaac ROS Dev container with image key x86_64.ros2_humble.user: /home/ashutosh/workspaces/isaac_ros-dev/
Building x86_64.ros2_humble.user base as image: isaac_ros_dev-x86_64
Building layered image for key x86_64.ros2_humble.user as isaac_ros_dev-x86_64
Using configured docker search paths: /home/ashutosh/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker
Checking if base image nvcr.io/isaac/ros:x86_64-ros2_humble-user_ac3bbc9a524b292030c0d582c20af435 exists on remote registry
Checking if base image nvcr.io/isaac/ros:x86_64-ros2_humble_45d368cdbbe4a484643464d0d492c764 exists on remote registry
Checking if base image nvcr.io/isaac/ros:x86_64_47e9206d1f6ce547682bb5e867dfadd0 exists on remote registry
Resolved the following 3 Dockerfiles for target image: x86_64.ros2_humble.user
/home/ashutosh/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/ashutosh/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble
/home/ashutosh/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64
Building /home/ashutosh/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64 as image: x86_64-image with base: 
[+] Building 6.4s (7/19)                                                                   docker:default
 => [internal] load build definition from Dockerfile.x86_64                                          0.0s
 => => transferring dockerfile: 5.97kB                                                               0.0s
 => [internal] load metadata for nvcr.io/nvidia/tritonserver:23.10-py3                               2.1s
 => [internal] load .dockerignore                                                                    0.0s
 => => transferring context: 2B                                                                      0.0s
 => [stage-0  1/16] FROM nvcr.io/nvidia/tritonserver:23.10-py3@sha256:83513cb05c7e53ada093cab904daa  0.0s
 => CACHED [stage-0  2/16] RUN mkdir -p /opt/nvidia/isaac_ros_dev_base && dpkg-query -W | sort > /o  0.0s
 => CACHED [stage-0  3/16] RUN --mount=type=cache,target=/var/cache/apt apt-get update && apt-get i  0.0s
 => ERROR [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac  4.3s
------                                                                                                    
 > [stage-0  4/16] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" /etc/apt/sources.list ||     echo "deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0" | tee -a /etc/apt/sources.list     && apt-get update:                                                                                               
0.120 Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
0.841 OK
0.886 deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0
1.241 Get:1 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy InRelease [3429 B]
2.072 Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease
2.072 Get:3 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]
2.248 Ign:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
2.296 Get:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages [29.8 kB]
2.296 Err:4 https://isaac.download.nvidia.com/isaac-ros/release-3 jammy/release-3.0 amd64 Packages
2.296   File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 192.229.232.31 443]
2.296   Hashes of expected file:
2.296    - Filesize:29756 [weak]
2.296    - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
2.296    - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
2.296    - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
2.296   Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
2.369 Get:5 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
3.565 Hit:6 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2204/x86_64  InRelease
3.629 Hit:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
3.708 Fetched 260 kB in 3s (93.8 kB/s)
3.708 Reading package lists...
4.248 W: https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
4.248 E: Failed to fetch https://isaac.download.nvidia.com/isaac-ros/release-3/dists/jammy/release-3.0/binary-amd64/Packages.gz  File has unexpected size (30010 != 29756). Mirror sync in progress? [IP: 192.229.232.31 443]
4.248    Hashes of expected file:
4.248     - Filesize:29756 [weak]
4.248     - SHA256:52e020ceb7e96ff56fe91d46c186552ba0f16eeed519c7e2f14c35b2b8ab2062
4.248     - SHA1:1eec0f5f970fa4757bf9eb7527880359d4084d7f [weak]
4.248     - MD5Sum:7e41bd8552c5ddd8a7292feacdbd4555 [weak]
4.248    Release file created at: Sat, 15 Jun 2024 03:14:19 +0000
4.248 E: Some index files failed to download. They have been ignored, or old ones used instead.
------
Dockerfile.x86_64:31
--------------------
[... cut ...]
ERROR: failed to solve: process "/bin/bash -c wget -qO - https://isaac.download.nvidia.com/isaac-ros/repos.key | apt-key add - &&     grep -qxF \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" /etc/apt/sources.list ||     echo \"deb https://isaac.download.nvidia.com/isaac-ros/release-3 $(lsb_release -cs) release-3.0\" | tee -a /etc/apt/sources.list     && apt-get update" did not complete successfully: exit code: 100
~/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts
`

HappySamuel, 2024-07-09:
Hi @liuyt , @ashutoshramola , @DailyNir 

Do you have solution for this issue? I am currently facing the same issue.

Best,
Samuel

ashutoshramola, 2024-07-10:
Not Yet. Still facing same issue