NVIDIA-ISAAC-ROS/isaac_ros_common #134: run_dev.sh failed on jetson orin nano while setting up for ROS realsense

Zeki411, 2024-07-14:
Hello, every body.

I'm trying to process Isaac ROS RealSense Setup on **Jetson Orin Nano** based on the link: https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html.
However, i had a problem while building the docker image using `cd ${ISAAC_ROS_WS}/src/isaac_ros_common && ./scripts/run_dev.sh -d ${ISAAC_ROS_WS}`, as follows:

>  
 240 |     # installed from source above.
 241 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 242 | >>>     mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
 243 | >>>     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble \
 244 | >>>     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash \
 245 | >>>     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary \
 246 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 247 | >>>     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary \
 248 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 249 | >>>     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 250 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 251 | >>>     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 252 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 253 | >>>     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 254 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 255 | >>>     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 256 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb
 257 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb" did not complete successfully: exit code: 2
/mnt/abko_ssd/workspaces/isaac_ros-dev/src/isaac_ros_common
 240 |     # installed from source above.
 241 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 242 | >>>     mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
 243 | >>>     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble \
 244 | >>>     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash \
 245 | >>>     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary \
 246 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 247 | >>>     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary \
 248 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 249 | >>>     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 250 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 251 | >>>     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 252 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 253 | >>>     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 254 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 255 | >>>     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 256 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb
 257 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb" did not complete successfully: exit code: 2
/mnt/abko_ssd/workspaces/isaac_ros-dev/src/isaac_ros_common
<

Does anyone know to handle this ? 
TIA

AntonJorg, 2024-07-15:
We have the same issue, would love to see it resolved!

torydebra, 2024-07-23:
It should be the same issue from here https://forums.developer.nvidia.com/t/error-building-image-in-orin-developer-kit-as-4gb-nano/299915