NVIDIA-ISAAC-ROS/isaac_ros_common #135: run-dev.sh fails with "Error 2"

delenius, 2024-07-15:
```
155.1 dh_auto_configure: error: cd .obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DCMAKE_INSTALL_PREFIX=/opt/ros/humble -DAMENT_PREFIX_PATH=/opt/ros/humble -DCMAKE_PREFIX_PATH=/opt/ros/humble -DBUILD_TESTING=OFF .. returned exit code 1
155.1 make[1]: *** [debian/rules:33: override_dh_auto_configure] Error 2
155.1 make[1]: Leaving directory '/opt/ros/humble/src/moveit_task_constructor/core'
155.1 make: *** [debian/rules:27: binary] Error 2
------

 1 warning found (use --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 17)
Dockerfile.ros2_humble:241
--------------------
 240 |     # installed from source above.
 241 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 242 | >>>     mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
 243 | >>>     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble \
 244 | >>>     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash \
 245 | >>>     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary \
 246 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 247 | >>>     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary \
 248 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 249 | >>>     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 250 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 251 | >>>     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 252 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 253 | >>>     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 254 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 255 | >>>     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 256 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb
 257 |
--------------------
ERROR: failed to solve: process "/bin/bash -c mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb" did not complete successfully: exit code: 2
~/workspaces/isaac_ros-dev
```

pedi, 2024-07-16:
The issue should be caused by moveit_task_constructor refactored their CMakelists recently. 

They introduced a new package `python_binding_tools` in [this commit](https://github.com/moveit/moveit_task_constructor/commit/29d12196dad3df012dd772b33d1ca94f0cfd7cff#diff-8f52846645b80806e11fbe318820659ca6d81d2fe305d815bb97f62d2102e1fdR15)

Since I'm not using the moveit capabilities in my project, I disabled the relevant code in the `Dockerfile.ros2_humble`. Though a better way would be to install the missing package first.

delenius, 2024-07-16:
@pedi which lines did you remove exactly?

delenius, 2024-07-16:
@pedi I submitted a PR that adds the dependency, and it works for me. Thanks for the suggestion!