NVIDIA-ISAAC-ROS/isaac_ros_common #139: Issue in Dockerfile.ros2_humble

Rnfudge02, 2024-07-23:
When building with docker using the command:

`./isaac_ros_common/scripts/build_image_layers.sh --image_key x86_64.ros2_humble.user.auv-x86_64 --build_arg USERNAME=auv-container --image_name auv-container:x86_64`

where auv-x86_64 is a dockerfile I have created, the following error (truncated) is thrown in the second stage of the build in the ros2_humble dockerfile:

`60.33 GNU C++17 (Ubuntu 11.4.0-1ubuntu1~22.04) version 11.4.0 (x86_64-linux-gnu)
60.33   compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version isl-0.24-GMP
60.33 
60.33 GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
60.33 ignoring duplicate directory "/usr/include/x86_64-linux-gnu/c++/11"
60.33 ignoring nonexistent directory "/usr/local/include/x86_64-linux-gnu"
60.33 ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/11/include-fixed"
60.33 ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/11/../../../../x86_64-linux-gnu/include"
60.33 #include "..." search starts here:
60.33 #include <...> search starts here:
60.33  /usr/include/c++/11
60.33  /usr/include/x86_64-linux-gnu/c++/11
60.33  /usr/include/c++/11/backward
60.33  /usr/lib/gcc/x86_64-linux-gnu/11/include
60.33  /usr/local/include
60.33  /usr/include/x86_64-linux-gnu
60.33  /usr/include
60.33 End of search list.
60.33 GNU C++17 (Ubuntu 11.4.0-1ubuntu1~22.04) version 11.4.0 (x86_64-linux-gnu)
60.33   compiled by GNU C version 11.4.0, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.1, isl version isl-0.24-GMP
60.33 
60.33 GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
60.33 Compiler executable checksum: d591828bb4d392ae8b7b160e5bb0b95f
60.33 COLLECT_GCC_OPTIONS='-g' '-O2' '-ffile-prefix-map=/opt/ros/humble/src/moveit_task_constructor/core=.' '-flto=auto' '-flto=auto' '-ffat-lto-objects' '-fstack-protector-strong' '-Wformat=1' '-D' 'NDEBUG' '-Wdate-time' '-D' '_FORTIFY_SOURCE=2' '-v' '-o' 'CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o' '-c' '-shared-libgcc' '-mtune=generic' '-march=x86-64' '-dumpdir' 'CMakeFiles/cmTC_45675.dir/'
60.33  as -v --gdwarf-5 --64 -o CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o /tmp/ccSUu3GK.s
60.33 GNU assembler version 2.38 (x86_64-linux-gnu) using BFD version (GNU Binutils for Ubuntu) 2.38
60.33 COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/
60.33 LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/11/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/local/cuda/lib64/stubs/:./:/usr/lib/gcc/x86_64-linux-gnu/11/../../../:/lib/:/usr/lib/
60.33 COLLECT_GCC_OPTIONS='-g' '-O2' '-ffile-prefix-map=/opt/ros/humble/src/moveit_task_constructor/core=.' '-flto=auto' '-flto=auto' '-ffat-lto-objects' '-fstack-protector-strong' '-Wformat=1' '-D' 'NDEBUG' '-Wdate-time' '-D' '_FORTIFY_SOURCE=2' '-v' '-o' 'CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o' '-c' '-shared-libgcc' '-mtune=generic' '-march=x86-64' '-dumpdir' 'CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.'
60.33 Linking CXX executable cmTC_45675
60.33 /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_45675.dir/link.txt --verbose=1
60.33 /usr/bin/c++ -g -O2 -ffile-prefix-map=/opt/ros/humble/src/moveit_task_constructor/core=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2   -v CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o -o cmTC_45675 
60.33 Using built-in specs.
60.33 COLLECT_GCC=/usr/bin/c++
60.33 COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
60.33 OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
60.33 OFFLOAD_TARGET_DEFAULT=1
60.33 Target: x86_64-linux-gnu
60.33 Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
60.33 Thread model: posix
60.33 Supported LTO compression algorithms: zlib zstd
60.33 gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
60.33 COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/
60.33 LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/11/:/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/11/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/local/cuda/lib64/stubs/:./:/usr/lib/gcc/x86_64-linux-gnu/11/../../../:/lib/:/usr/lib/
60.33 COLLECT_GCC_OPTIONS='-g' '-O2' '-ffile-prefix-map=/opt/ros/humble/src/moveit_task_constructor/core=.' '-flto=auto' '-flto=auto' '-ffat-lto-objects' '-fstack-protector-strong' '-Wformat=1' '-D' 'NDEBUG' '-Wdate-time' '-D' '_FORTIFY_SOURCE=2' '-v' '-o' 'cmTC_45675' '-shared-libgcc' '-mtune=generic' '-march=x86-64' '-dumpdir' 'cmTC_45675.'
60.33  /usr/lib/gcc/x86_64-linux-gnu/11/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/11/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper -plugin-opt=-fresolution=/tmp/cc2hVNWt.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -flto=auto -flto=auto --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -z relro -o cmTC_45675 /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/Scrt1.o /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/11/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/11 -L/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/11/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/local/cuda/lib64/stubs -L. -L/usr/lib/gcc/x86_64-linux-gnu/11/../../.. CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /usr/lib/gcc/x86_64-linux-gnu/11/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crtn.o
60.33 /usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper -fresolution=/tmp/cc2hVNWt.res -flinker-output=pie CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o 
60.33 /usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper -fresolution=/tmp/cc2hVNWt.res -flinker-output=pie CMakeFiles/cmTC_45675.dir/CMakeCXXCompilerABI.cpp.o 
60.33 LTO parallelism level set to 16
60.33 /usr/bin/c++ @/tmp/cc8ly5kG
60.33 Using built-in specs.
60.33 COLLECT_GCC=/usr/bin/c++
60.33 OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
60.33 OFFLOAD_TARGET_DEFAULT=1
60.33 Target: x86_64-linux-gnu
60.33 Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-c [... cut ...] e rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb" did not complete successfully: exit code: 2`

rob-pointcloud, 2024-07-24:
hello, just to second this, I am getting the same error. I am running on Jetson Orin nano, and only using the base docker, not adding anything. Here is how I am running the command:

`./run_dev.sh -d /ssd/workspace/src/isaac_ros-dev/`

tuannv-github, 2024-07-25:
I got the same error when running ` ./run_dev.sh` on Ubuntu 22.04 x86_64 machine.

> ERROR: failed to solve: process "/bin/bash -c mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble     && cd [...]
~/isaac_ros_common/scripts

hemalshahNV, 2024-07-25:
It looks like there was a change committed to the MoveIt repositories a couple of weeks ago that may be causing this. We're working on reproducing locally and see if we can anchor a version to resolve.

Separately, you shouldn't need to rebuild these Docker images unless you've modified Dockerfile.x86_64, Dockerfile.aarch64, and/or Dockerfile.ros2_humble. Just to confirm, have any of you modified these files and then run run_dev.sh or build_image_layers.sh or running unmodified (which should have picked up the cached prebuilt images available on NVIDIA Container Registry).

tuannv-github, 2024-07-25:
I do not modify anything; I just run the script. I cannot find instructions for pulling and using cached prebuilt images. Is it available online?

DJT777, 2024-07-25:
@hemalshahNV 

I am just trying to run run_dev.sh from CLI from a fresh clone of the repository and getting the same error:

```
67.63 make[1]: Leaving directory '/opt/ros/humble/src/moveit_task_constructor/core'
67.63 make: *** [debian/rules:27: binary] Error 2
------

 1 warning found (use --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 17)
Dockerfile.ros2_humble:241
--------------------
 240 |     # installed from source above.
 241 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 242 | >>>     mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src \
 243 | >>>     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble \
 244 | >>>     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash \
 245 | >>>     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary \
 246 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 247 | >>>     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary \
 248 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 249 | >>>     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 250 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 251 | >>>     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 252 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 253 | >>>     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 254 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb \
 255 | >>>     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck \
 256 | >>>     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb
 257 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c mkdir -p ${ROS_ROOT}/src && cd ${ROS_ROOT}/src     && git clone https://github.com/ros-planning/moveit_task_constructor.git -b humble     && cd moveit_task_constructor && source ${ROS_ROOT}/setup.bash     && cd msgs && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd rviz_marker_tools && bloom-generate rosdebian && fakeroot debian/rules binary     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd core && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd capabilities && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd visualization && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb     && cd demo && bloom-generate rosdebian && fakeroot debian/rules binary DEB_BUILD_OPTIONS=nocheck     && cd ../ && apt-get install -y ./*.deb && rm ./*.deb" did not complete successfully: exit code: 2
~/repos/isaac_ros_common
```

Rnfudge02, 2024-07-25:
Sorry, I did not mean to close this issue

DJT777, 2024-07-25:
Looks like someone made a PR that has this issue resolved:

Originally posted by @delenius in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/issues/135#issuecomment-2231278123_

Rnfudge02, 2024-07-26:
I tested the change @delenius suggested in the PR by inserting the code at line 238 in Dockerfile.ros2_humble, and can confirm it resolves the issue. However, when using build_image_layers.sh it did not pull a pre-built image, despite no changes being made in the original dockerfiles.