NVIDIA-ISAAC-ROS/isaac_ros_common #141: Roadmap for ROS2 Jazzy

JohannesPankert, 2024-07-29:
Hello
do you have a release for Ubuntu 24.04 and ROS2 Jazzy on your roadmap? If so, when do you plan to release it?
Cheers
Johannes

brennand, 2025-04-17:
I agree, after 1 year of jazzy being out it seems like it should be supported.

hemalshahNV, 2025-04-17:
Yes, very much on our roadmap and working on it now. Jetpack will soon be on Ubuntu 24.04 and then we can live the Jazzy life.

tomasz-lewicki, 2025-04-22:
Hi Hemal,

Do you have an estimated timeline for the release? Based on the Jetson roadmap, it looks like Ubuntu 24.04 support via JetPack is expected Q3 2025. Should we assume ROS 2 Jazzy support will follow that same schedule?

https://developer.nvidia.com/embedded/develop/roadmap

Thanks,
Tomasz