NVIDIA-ISAAC-ROS/isaac_ros_common #147: The script `run_dev.sh` runs failed with urlopen error.

otischung, 2024-09-22:
I ran the `isaac_ros_commom/scripts/run_dev.sh` and got the following error:

```
ubuntu@jetson02:~/workspaces/isaac_ros_common/scripts$ ./run_dev.sh -d /home/ubuntu/workspaces
Launching Isaac ROS Dev container with image key aarch64.ros2_humble.user: /home/ubuntu/workspaces
Building aarch64.ros2_humble.user base as image: isaac_ros_dev-aarch64
Building layered image for key aarch64.ros2_humble.user as isaac_ros_dev-aarch64
Using configured docker search paths: /home/ubuntu/workspaces/isaac_ros_common/scripts/../docker
Checking if base image nvcr.io/isaac/ros:aarch64-ros2_humble-user_763d6cb6c3f1162f1adbfeba5f7569e4 exists on remote registry
Checking if base image nvcr.io/isaac/ros:aarch64-ros2_humble_de2adbd690fb9b0f8d1c468d64030182 exists on remote registry
Checking if base image nvcr.io/isaac/ros:aarch64_9f6b6a850299a7d0ea73d257f66b2850 exists on remote registry
Resolved the following 3 Dockerfiles for target image: aarch64.ros2_humble.user
/home/ubuntu/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/ubuntu/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble
/home/ubuntu/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.aarch64
Building /home/ubuntu/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.aarch64 as image: aarch64-image with base:
[+] Building 0.4s (26/26) FINISHED                           docker:default
 => [internal] load build definition from Dockerfile.aarch64           0.0s
 => => transferring dockerfile: 8.30kB                                 0.0s
 => [internal] load metadata for nvcr.io/nvidia/l4t-cuda:12.2.12-deve  0.4s
 => [internal] load .dockerignore                                      0.0s
 => => transferring context: 2B                                        0.0s
 => [stage-0  1/22] FROM nvcr.io/nvidia/l4t-cuda:12.2.12-devel@sha256  0.0s
 => CACHED [stage-0  2/22] RUN mkdir -p /opt/nvidia/isaac_ros_dev_bas  0.0s
 => CACHED [stage-0  3/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  4/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  5/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  6/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  7/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  8/22] RUN update-alternatives --install /usr/bin  0.0s
 => CACHED [stage-0  9/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 10/22] RUN python3 -m pip install -U     Cython    0.0s
 => CACHED [stage-0 11/22] RUN update-alternatives --install /usr/bin  0.0s
 => CACHED [stage-0 12/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 13/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 14/22] RUN mkdir -p /opt/nvidia/tao && cd /opt/nv  0.0s
 => CACHED [stage-0 15/22] RUN python3 -m pip install --no-cache       0.0s
 => CACHED [stage-0 16/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 17/22] RUN  --mount=type=cache,target=/var/cache/  0.0s
 => CACHED [stage-0 18/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 19/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 20/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0 21/22] RUN python3 -m pip install -U     jetson-s  0.0s
 => CACHED [stage-0 22/22] RUN mkdir -p /opt/nvidia/isaac_ros_dev_bas  0.0s
 => exporting to image                                                 0.0s
 => => exporting layers                                                0.0s
 => => writing image sha256:ff925fca5946bea43b48af5a395616dc26b48f0d1  0.0s
 => => naming to docker.io/library/aarch64-image                       0.0s
Building /home/ubuntu/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble as image: ros2_humble-image with base: aarch64-image
[+] Building 6.6s (14/26)                                    docker:default
 => [internal] load build definition from Dockerfile.ros2_humble       0.0s
 => => transferring dockerfile: 13.51kB                                0.0s
 => WARN: InvalidDefaultArgInFrom: Default value for ARG $BASE_IMAGE   0.0s
 => WARN: LegacyKeyValueFormat: "ENV key=value" should be used instea  0.0s
 => [internal] load metadata for docker.io/library/aarch64-image:late  0.0s
 => [internal] load .dockerignore                                      0.0s
 => => transferring context: 2B                                        0.0s
 => [stage-0  1/22] FROM docker.io/library/aarch64-image:latest        0.0s
 => [internal] load build context                                      0.0s
 => => transferring context: 155B                                      0.0s
 => CACHED [stage-0  2/22] RUN mkdir -p /opt/nvidia/isaac_ros_dev_bas  0.0s
 => CACHED [stage-0  3/22] RUN locale-gen en_US en_US.UTF-8 && update  0.0s
 => CACHED [stage-0  4/22] RUN echo "Warning: Using the PYTHONWARNING  0.0s
 => CACHED [stage-0  5/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  6/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  7/22] RUN python3 -m pip install -U         flak  0.0s
 => CACHED [stage-0  8/22] RUN --mount=type=cache,target=/var/cache/a  0.0s
 => CACHED [stage-0  9/22] COPY rosdep/extra_rosdeps.yaml /etc/ros/ro  0.0s
 => ERROR [stage-0 10/22] RUN --mount=type=cache,target=/var/cache/ap  6.5s
------
 > [stage-0 10/22] RUN --mount=type=cache,target=/var/cache/apt     rosdep init     && echo "yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list     && rosdep update:
0.986 Wrote /etc/ros/rosdep/sources.list.d/20-default.list
0.986 Recommended: please run
0.986
0.986   rosdep update
0.986
1.112 yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml
1.899 Warning: running 'rosdep update' as root is not recommended.
1.899   You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
3.614 ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml]:
3.614   <urlopen error [Errno 104] Connection reset by peer> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml)
3.705 ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml]:
3.705   <urlopen error [Errno 104] Connection reset by peer> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml)
6.268 ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
6.268   Failed to download target platform data for gbpdistro:
6.268   <urlopen error [Errno 104] Connection reset by peer>
6.366 ERROR: error loading sources list:
6.366   <urlopen error <urlopen error [Errno 104] Connection reset by peer> (https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml)>
6.367 reading in sources list data from /etc/ros/rosdep/sources.list.d
6.367 Hit file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml
6.367 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
6.367 Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
6.367 Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
------
Dockerfile.ros2_humble:130
--------------------
 129 |     COPY rosdep/extra_rosdeps.yaml /etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml
 130 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 131 | >>>     rosdep init \
 132 | >>>     && echo "yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list \
 133 | >>>     && rosdep update
 134 |
--------------------
ERROR: failed to solve: process "/bin/bash -c rosdep init     && echo \"yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml\" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list     && rosdep update" did not complete successfully: exit code: 1
~/workspaces/isaac_ros_common/scripts
```

The git repo commit:
```
commit e8760a63dd9f4e428dbe01b597f265020792217c (HEAD -> main, origin/main, origin/HEAD)
Merge: 77777d3 5c9000a
Author: Jaiveer Singh <84546269+jaiveersinghNV@users.noreply.github.com>
Date:   Wed Jul 31 12:15:33 2024 -0700

    Merge pull request #142 from NVIDIA-ISAAC-ROS/hotfix-release-3.0-2

    Add pybindingtools dependency for MoveIt Task Constructor
````

Platform System: Jetpack 6.0 [L4T 36.3.0]

otischung, 2024-09-22:
Same as in the x86-64 platform

```
1 warning found (use docker --debug to expand):                                                                                                             - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 20)                                                  Building /home/otischung/workspaces/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble as image: ros2_humble-image with base: x86_64-image           [+] Building 229.2s (14/26)                                                                              docker:default                                      => [internal] load build definition from Dockerfile.ros2_humble                                                   0.0s                                      => => transferring dockerfile: 13.51kB                                                                            0.0s                                      => WARN: InvalidDefaultArgInFrom: Default value for ARG $BASE_IMAGE results in empty or invalid base image name   0.0s                                      => WARN: LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 17)  0.0s                                      => [internal] load metadata for docker.io/library/x86_64-image:latest                                             0.0s                                      => [internal] load .dockerignore                                                                                  0.0s                                      => => transferring context: 2B                                                                                    0.0s                                      => [stage-0  1/22] FROM docker.io/library/x86_64-image:latest                                                     0.3s                                      => [internal] load build context                                                                                  0.1s                                      => => transferring context: 276.31kB                                                                              0.0s                                      => [stage-0  2/22] RUN mkdir -p /opt/nvidia/isaac_ros_dev_base && dpkg-query -W | sort > /opt/nvidia/isaac_ros_d  0.1s                                      => [stage-0  3/22] RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8          1.4s                                      => [stage-0  4/22] RUN echo "Warning: Using the PYTHONWARNINGS environment variable to silence setup.py and easy  0.1s                                      => [stage-0  5/22] RUN --mount=type=cache,target=/var/cache/apt     curl -sSL https://raw.githubusercontent.com/  4.1s                                      => [stage-0  6/22] RUN --mount=type=cache,target=/var/cache/apt apt-get update && apt-get install -y         de  77.1s                                      => [stage-0  7/22] RUN python3 -m pip install -U         flake8-blind-except         flake8-builtins         fl  15.1s                                      => [stage-0  8/22] RUN --mount=type=cache,target=/var/cache/apt apt-get update && apt-get install -y     ros-h  130.1s                                      => [stage-0  9/22] COPY rosdep/extra_rosdeps.yaml /etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml                0.0s                                      => ERROR [stage-0 10/22] RUN --mount=type=cache,target=/var/cache/apt     rosdep init     && echo "yaml file:///  0.6s                                     ------
 > [stage-0 10/22] RUN --mount=type=cache,target=/var/cache/apt     rosdep init     && echo "yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list     && rosdep update:
0.497 ERROR: cannot download default sources list from:
0.497 https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list                                                          0.497 Website may be down.
0.497 <urlopen error <urlopen error [Errno 104] Connection reset by peer> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list)>
------
Dockerfile.ros2_humble:130
--------------------
 129 |     COPY rosdep/extra_rosdeps.yaml /etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml                                                                   130 | >>> RUN --mount=type=cache,target=/var/cache/apt \
 131 | >>>     rosdep init \
 132 | >>>     && echo "yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list \
 133 | >>>     && rosdep update
 134 |
--------------------
ERROR: failed to solve: process "/bin/bash -c rosdep init     && echo \"yaml file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml\" | tee /etc/ros/rosdep/sources.list.d/00-nvidia-isaac.list     && rosdep update" did not complete successfully: exit code: 4
~/workspaces/isaac_ros_common/scripts
```

Platform system: Linux 5.15.0-121-generic #131~20.04.1-Ubuntu, NVIDIA-SMI 555.42.06, Driver Version: 555.42.06, CUDA Version: 12.5

j3soon, 2024-09-22:
Hi,

I wasn't able to reproduce the issue on my x86_64 machine. I suspect it might be related to your internet connection. Could you try connecting to a different network and see if the problem persists?

In addition, on your current network, please check if you're able to access the following URLs:
* https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
* https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list

otischung, 2024-09-24:
Since you couldn't reproduce this error, I tried a different network and repeated the same steps. The image was successfully built. Then, I used a VPN on the same target machine via the same wired connection, and it worked. It seems that our IPs were being rate-limited by GitHub. I've found a solution for this issue and will now close it. I really appreciate your comment.

closed 2024-09-24 by otischung