NVIDIA-ISAAC-ROS/isaac_ros_common #155: build_image_layers.sh: Default value for ARG ${BASE_IMAGE} results in empty or invalid base image name

delenius, 2024-11-08:
When I run `./run-dev.sh` (which subsequently runs `./build_image_layers.sh`), there is an error about the `BASE_IMAGE` build arg:

```
Building /homes/elenius/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble as image: ros2_humble-image with base: x86_64-image
BASE_IMAGE_ARG is --build-arg BASE_IMAGE=x86_64-image
[+] Building 0.1s (27/27) FINISHED                                                                                                                          docker:default
 => [internal] load build definition from Dockerfile.ros2_humble                                                                                                      0.0s
 => => transferring dockerfile: 13.71kB                                                                                                                               0.0s
 => WARN: InvalidDefaultArgInFrom: Default value for ARG $BASE_IMAGE results in empty or invalid base image name (line 10)                                            0.0s
 => WARN: LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 17)                                                     0.0s
 => [internal] load metadata for docker.io/library/x86_64-image:latest                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                                     0.0s
 => => transferring context: 2B                                                                                                                                       0.0s
 => [internal] load build context                                                                                                                                     0.0s
 => => transferring context: 276.05kB                                                                                                                                 0.0s
 => [stage-0  1/22] FROM docker.io/library/x86_64-image:latest                                                                                                        0.0s
```
(and similarly for subsequent layers). Note that I added an echo of `BASE_IMAGE_ARG` in the script, the output of which is shown in the output above.
Strangely enough, it still seems to find the right base image, even though there is no default value set in the Dockerfile.
This machine has docker version Docker version 27.3.1, build ce12230

When I run the same exact thing on another machine with docker Docker version 26.1.4, build 5650f9b, these warnings do not appear!

hemalshahNV, 2024-11-08:
Docker recently began being more insistent on the key value format. The BASE_IMAGE default argument is only really used for Dockerfile.x86_64 and Dockerfile.aarch64. For all of the remaining Dockerfiles for the downstream image layers, build_image_layers.sh feeds one image into the next using that argument, so its default value is never used.

delenius, 2024-11-08:
Oh, the `ENV key=value` warning is unrelated. For example, for the last stage, I get this output, without that warning:
```
Building /homes/elenius/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user as image: isaac_ros_dev-x86_64 with base: ros2_humble-image
BASE_IMAGE_ARG is --build-arg BASE_IMAGE=ros2_humble-image
[+] Building 0.1s (17/17) FINISHED                                                                                                                       docker:default
 => [internal] load build definition from Dockerfile.user                                                                                                          0.0s
 => => transferring dockerfile: 2.36kB                                                                                                                             0.0s
 => WARN: InvalidDefaultArgInFrom: Default value for ARG ${BASE_IMAGE} results in empty or invalid base image name (line 10)                                       0.0s
 => [internal] load metadata for docker.io/library/ros2_humble-image:latest                                                                                        0.0s
 => [internal] load .dockerignore                                                                                                                                  0.0s
 => => transferring context: 2B                                                                                                                                    0.0s
 => [stage-0  1/12] FROM docker.io/library/ros2_humble-image:latest                                                                                                0.0s
```
Is it just that docker doesn't like not having a default for ARGs now?

closed 2026-02-02 by jaiveersinghNV