NVIDIA-ISAAC-ROS/isaac_ros_common #158: How do I use an older version of the IsaacROSDev Container?

brettpac, 2024-11-21:
Hi @hemalshahNV 
A recent set of application demos found here: https://github.com/robosoft-ai/nova_carter_sm_library/tree/main

A bug appeared in the latest version (11/5) of the IsaacROSDev Container.
https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/containers/ros

I don't have a 100% clear vision of why the bug is occuring (the problem only affects a few states) but it seems to be related to the RMW implementation and the canceling of action servers.

I normally launch the IRDC (IsaacROSDev Container) with the following command:
./src/isaac_ros_common/scripts/run_dev.sh  -d  ~/workspace/humble_ws/

How can I specify for the run_dev.sh to run the 9/30 container version? 
tag: nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_92060e94476d1fc5d11171f6ed37f110

Thank you,

hemalshahNV, 2024-11-21:
Interesting! We need to figure out what happened here with the latest ROS 2 Humble sync then.

You can forcefully move back by creating your own file called Dockerfile.x86_64.ros2_humble and having as its first line FROM: nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_92060e94476d1fc5d11171f6ed37f110. run_dev.sh/build_image_layers.sh will find this "composite key" Dockerfile first and run from there rather than try to calculate anything.

brettpac, 2024-12-13:
Finally solved the bug.Ended up being some trivial setting in IsaacSim. 
But, I just wanted to say that the technique of described in @hemalshahNV's last comment worked flawlessly. I'd like to suggest that this issue be pinned (so it appears in a bar that goes across the top of the page) to the issues page once closed for future users.

Thank you, issue is resolved.