NVIDIA-ISAAC-ROS/isaac_ros_common #161: run_dev.sh aarch64 Dockerfile: l4t-repo.nvidia.com domain cannot be resolved when adding internal key

edfus, 2024-12-10:
run_dev.sh: Blocking Issue: Unable to Fetch Keys from NVIDIA L4T Repo in Aarch64 Dockerfile

I’m encountering a critical issue with the Aarch64 Dockerfile. Specifically, the line:

apt-key adv --fetch-keys http://l4t-repo.nvidia.com/jetson-ota-internal.key

fails with a “no name” error. This appears to happen because the URL for the L4T repository key (http://l4t-repo.nvidia.com/jetson-ota-internal.key) is inaccessible.

Setup Details:
	•	Hardware: Jetson AGX Orin 64G
	•	Software: JetPack R36.4

This issue is currently blocking my progress, as I am unable to proceed with my build until the repository key can be fetched. Any guidance or fixes would be greatly appreciated. Thank you in advance for your support!

SnoopyGeeek, 2024-12-11:
Whooa. That is exactly the same problem I am facing. Can anyone shed some light on it?

edfus, 2024-12-11:
Used Google’s Dig tool to check l4t-repo.nvidia.com, and it returned a “records not found” error. Is the domain intended to be an internal NVIDIA Corporation domain and not publicly accessible? Below is the output:

```
id 63944
opcode QUERY
rcode NOERROR
flags QR RD RA
;QUESTION
l4t-repo.nvidia.com. IN A
;ANSWER
;AUTHORITY
nvidia.com. 900 IN SOA dns2.nvidia.com. hostmaster.nvidia.com. 2017027456 10800 3600 2419200 900
;ADDITIONAL
```

edfus, 2024-12-11:
@hemalshahNV, I would greatly appreciate your assistance regarding the issue with the apt-key adv --fetch-keys command failing due to the inaccessibility of l4t-repo.nvidia.com. Upon further investigation, I performed a DNS lookup for l4t-repo.nvidia.com using Google’s Dig tool and received the following output:

```
id 63944
opcode QUERY
rcode NOERROR
flags QR RD RA
;QUESTION
l4t-repo.nvidia.com. IN A
;ANSWER
;AUTHORITY
nvidia.com. 900 IN SOA dns2.nvidia.com. hostmaster.nvidia.com. 2017027456 10800 3600 2419200 900
;ADDITIONAL
```

This indicates that no A record was provided for l4t-repo.nvidia.com. The presence of an SOA (Start of Authority) record for nvidia.com in the authority section suggests that while the domain exists, it may not be configured for public access.

Could you please confirm whether l4t-repo.nvidia.com is intended to be publicly accessible? If it is, any guidance on how to resolve this issue would be highly valuable, as it is currently hindering my progress.

Thank you in advance for your support.

hemalshahNV, 2024-12-13:
Hi, @edfus . Yes, l4t-repo.nvidia.com is not meant to be publicly accessible which we use for internal, pre-release builds of Jetpack for testing purposes. Isaac 3.2 should have this resolved but we will check again. The workaround would be to edit /etc/apt/sources.list.d/ and remove the entries for l4t-repo.nvidia.com in the meantime.

edfus, 2024-12-21:
The issue is resolved with the Isaac 3.2 release. Thanks again for your support @hemalshahNV

closed 2024-12-21 by edfus