NVIDIA-ISAAC-ROS/isaac_ros_common #167: run_dev.sh exits without catching git-lfs error

mschweig, 2025-01-03:
Hi, 

the git-lfs installation check in `run_dev.sh` does not work if git is installed but without git-lfs (which is default on Ubuntu 22.04)

`run_dev.sh` exits without catching the missing installation of git-lfs. 

[image]

I already opened PR #166 which fixes this issue. 

Thanks

EnriqueSolarte, 2025-07-23:
I tried to build a Docker image using run_dev.sh. Interesting. the version at `c4ba19faa7b302cdc76ff04f45d3a630279e9cba` does not catch `git lfs &>/dev/null` error. 
I run the following command but not error nor docker building process was executed. 
```bash 
./run_dev.sh -d ${ISAAC_ROS_WS}
```
Turns out `git-lfs` lib was missing
```
sudo apt-get install git-lfs
```
for context reference, my device is: 
```
Jetson Orin Nano
# R36 (release), REVISION: 4.4, GCID: 41062509, BOARD: generic, EABI: aarch64, DATE: Mon Jun 16 16:07:13 UTC 2025
# KERNEL_VARIANT: oot
TARGET_USERSPACE_LIB_DIR=nvidia
TARGET_USERSPACE_LIB_DIR_PATH=usr/lib/aarch64-linux-gnu/nvidia
```

jaiveersinghNV, 2025-07-24:
Git LFS is a requirement to use Isaac ROS, as indicated in our Developer Environment Setup docs:
https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html#developer-environment-setup

That said, I'm surprised that our `run_dev.sh` script is not appropriately detecting that Git LFS is missing. We'll track that as a bug to fix in an upcoming release.