NVIDIA-ISAAC-ROS/isaac_ros_common #168: build_image_layers.sh Error on common 9/18

RoboSubUWF, 2025-01-07:
I am trying to run the `sudo ./build_image_layers.sh -i aarch` command and I keep receiving this error on common 9/18:

`ERROR [common  9/18] RUN python3 -m pip install -U         argcompl  181.7s 
179.4 Failed to build mapbox_earcut
179.4 ERROR: Could not build wheels for mapbox_earcut, which is required to install pyproject.toml-based projects
------

 1 warning found (use docker --debug to expand):
 - UndefinedVar: Usage of undefined variable '$CMAKE_PREFIX_PATH' (line 267)
Dockerfile.aarch64:150
--------------------
 149 |     # Python3 (PIP)
 150 | >>> RUN python3 -m pip install -U \
 151 | >>>         argcomplete \
 152 | >>>         autopep8 \
 153 | >>>         Cython \
 154 | >>>         flake8 \
 155 | >>>         flake8-blind-except \
 156 | >>>         flake8-builtins \
 157 | >>>         flake8-class-newline \
 158 | >>>         flake8-comprehensions \
 159 | >>>         flake8-deprecated \
 160 | >>>         flake8-docstrings \
 161 | >>>         flake8-import-order \
 162 | >>>         flake8-quotes \
 163 | >>>         gpustat==0.6.0 \
 164 | >>>         importlib_resources \
 165 | >>>         networkx \
 166 | >>>         ninja \
 167 | >>>         "numpy>=1.24.4,<2" \
 168 | >>>         numpy-quaternion \
 169 | >>>         onnx \
 170 | >>>         pydocstyle \
 171 | >>>         pymongo \
 172 | >>>         pyyaml \
 173 | >>>         "scipy>=1.7.0" \
 174 | >>>         scikit-image \
 175 | >>>         scikit-learn \
 176 | >>>         "setuptools_scm>=6.2" \
 177 | >>>         tqdm \
 178 | >>>         trimesh \
 179 | >>>         "warp-lang>=0.9.0" \
 180 | >>>         wheel \
 181 | >>>         "yourdfpy>=0.0.53"
 182 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c python3 -m pip install -U         argcomplete         autopep8         Cython         flake8         flake8-blind-except         flake8-builtins         flake8-class-newline         flake8-comprehensions         flake8-deprecated         flake8-docstrings         flake8-import-order         flake8-quotes         gpustat==0.6.0         importlib_resources         networkx         ninja         \"numpy>=1.24.4,<2\"         numpy-quaternion         onnx         pydocstyle         pymongo         pyyaml         \"scipy>=1.7.0\"         scikit-image         scikit-learn         \"setuptools_scm>=6.2\"         tqdm         trimesh         \"warp-lang>=0.9.0\"         wheel         \"yourdfpy>=0.0.53\"" did not complete successfully: exit code: 1`

I am running on a Jetson Orin Nano Jetpack 6.1. Does anyone know a solution to this?

shawnhanna, 2025-01-07:
I also see this same error

shawnhanna, 2025-01-08:
Nothing scientific in my fix here, but my guess is that it's a pip/dependency issue. I got this to work by separating the pip installs into separate commands like so (I imagine this could be simplified better but I'm not spending more time on this fix):

```
# Python3 (PIP)
RUN python3 -m pip install -U \
        argcomplete \
        autopep8 \
        Cython \
        flake8 \
        flake8-blind-except \
        flake8-builtins \
        flake8-class-newline \
        flake8-comprehensions \
        flake8-deprecated \
        flake8-docstrings \
        flake8-import-order \
        flake8-quotes \
        gpustat==0.6.0 \
        importlib_resources \
        networkx
RUN python3 -m pip install -U \
        ninja \
        "numpy>=1.24.4,<2" \
        numpy-quaternion
RUN python3 -m pip install -U \
        onnx \
        pydocstyle \
        pymongo \
        pyyaml
RUN python3 -m pip install -U \
        "scipy>=1.7.0"
RUN python3 -m pip install -U \
        scikit-image
RUN python3 -m pip install -U \
        scikit-learn
RUN python3 -m pip install -U \
        "setuptools_scm>=6.2" \
        tqdm \
        trimesh
RUN python3 -m pip install -U \
        "warp-lang>=0.9.0" \
        wheel \
        "yourdfpy>=0.0.53"
```

RoboSubUWF, 2025-01-08:
Changing the code to your solution worked for me, but then I received a warning right as the `sudo ./build_image_layers.sh -i aarch64` command was about to finish running saying: 
` => => naming to docker.io/library/aarch64-image                           0.0s

 1 warning found (use docker --debug to expand):
 - UndefinedVar: Usage of undefined variable '$CMAKE_PREFIX_PATH' (line 274)
`

I tried changing the current line 274 as shown below: 
274 Original: `ENV CMAKE_PREFIX_PATH=$CUDSS_DIR:$CMAKE_PREFIX_PATH`
274 New: `ENV CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH:-default/path}:$CUDSS_DIR`

But still received the same error. I also had an error at step 8/12 but fixed it by replacing lines 370-379 with 
`RUN --mount=type=cache,target=/var/cache/apt \
    apt-get update && apt-get install -y libboost-dev"` 
  and this fixed the issue
  
  Did you happen to get this error at line 274 as well?

ba2sakal, 2025-01-09:
I added these 2 lines before the PIP3 installations

### Install ninja-build - MANUAL EDIT
RUN apt-get update && apt-get install -y ninja-build

RUN python3 -m pip install -U mapbox_earcut
###

dirksavage88, 2025-04-04:
I separated out the pip installs, and my build on Jetpack 6 seems to fail on this line:

`--------------------
ERROR: failed to solve: process "/bin/bash -c python3 -m pip install -U yourdfpy" did not complete successfully: exit code: 1
~/workspaces/isaac_ros-dev/ros_ws/isaac_ros_common
`