NVIDIA-ISAAC-ROS/isaac_ros_common #180: ZED Explorer stuck on "Waiting for Camera" with ZED Mini

MatheusOPP, 2025-03-07:
I have been trying to set up Isaac ROS to run pose estimation with [FoundationPose](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_pose_estimation/isaac_ros_foundationpose/index.html) while getting a video feed from a ZED Mini Camera on a Ubuntu 22.04 x86_64 machine.

For this sake, I have followed the instructions on setting up the [development environment](https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html) and for the [ZED setup](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/zed_setup.html). I get no particular errors or issues when doing this, but when running ZED Explorer by the end of the ZED setup instructions, I run into issues.

At first, I had the following issue impeding ZED Explorer from opening any windows:

```
qt.qpa.xcb: could not connect to display :1
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
```

Which was solved by running `xhost +local:docker` in the host machine. However, now ZED Explorer is stuck in the following screen, with a "Waiting for Camera" message that never goes away.

[image]

To troubleshoot this, I tried installing the ZED SDK on my host machine (outside of the Docker container), and it works perfectly. The video feed is also able to be captured by applications such as Cheese. So this is definitely not a hardware/camera issue nor an issue in the host machine.

Furthermore, it seems that the Docker container is able to detect the camera and video feed, as show by the outputs `ls /dev/video*` and `lsusb` from inside the Docker container. Both of those outputs are also the same in the host machine.

```
admin@matheusopp-MS-7E34:/workspaces/isaac_ros-dev$ ls /dev/video*
/dev/video0  /dev/video1
```

```
admin@matheusopp-MS-7E34:/workspaces/isaac_ros-dev$ lsusb | grep ZED
Bus 002 Device 003: ID 2b03:f682 STEREOLABS ZED-M camera
Bus 001 Device 012: ID 2b03:f681 STEREOLABS ZED-M HID Interface
```

I would appreciate any help in trying to debug/troubleshoot this issue.

christian-helms, 2025-03-09:
I just bumped into (probably) the exact same problem. Modifying [run_dev.sh](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/scripts/run_dev.sh) to have a proper communication setup between X-Server and docker as described in [How to Install ZED with Docker on Linux](https://www.stereolabs.com/docs/docker/install-guide-linux) fixed it for me. Concretely, putting `xhost +si:localuser:root` before the docker run command and adding the two options `-e Display -v /tmp/.X11-unix:/tmp/.X11-unix` solves my problem.

```
xhost +si:localuser:root

docker run -it --rm \
    --privileged \
    --network host \
    --ipc=host \
    ${DOCKER_ARGS[@]} \
    -v $ISAAC_ROS_DEV_DIR:/workspaces/isaac_ros-dev \
    -v /etc/localtime:/etc/localtime:ro \
    --name "$CONTAINER_NAME" \
    --runtime nvidia \
    --entrypoint /usr/local/bin/scripts/workspace-entrypoint.sh \
    --workdir /workspaces/isaac_ros-dev \
    $BASE_NAME \
    /bin/bash
    -e DISPLAY 
    -v /tmp/.X11-unix:/tmp/.X11-unix```

My setup aligns aligns with what you have mentioned and am actually also setting up FoundationPose :)

MatheusOPP, 2025-03-11:
@christian-helms Did this fix the issue with the camera feed/video not showing in ZED Explorer for you? I added those lines to `run_dev.sh` and they certainly fix the display issue, but unfortunately I still get a "Waiting for Camera" message when opening ZED Explorer and no camera feed at all...

MatheusOPP, 2025-03-11:
Actually, never mind. I tried to troubleshoot the issue by getting the `dev/video0` output with FFMPEG, and this was the output.

```
admin@matheusopp-MS-7E34:/workspaces/isaac_ros-dev$ ffmpeg -f v4l2 -i /dev/video0 -f x11grab -s 640x480 -r 30 -t 10 output.mkv
[...]
[video4linux2,v4l2 @ 0x60afccd9bd00] Cannot open video device /dev/video0: Permission denied
/dev/video0: Permission denied
```

Which made me suspicious of permission issues. So I just ran ZED Explorer with `sudo` and now it works.

closed 2025-03-11 by MatheusOPP