NVIDIA-ISAAC-ROS/isaac_ros_common #183: run_dev python packages issues

arthurgritzky, 2025-04-05:
I am trying to run ISAAC Ros on a Jetson board.
I set everything up according to the set up guide. 
When running  cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \ 
./scripts/run_dev.sh
I run into python package issues:

1 warning found (use docker --debug to expand):
 - UndefinedVar: Usage of undefined variable '$CMAKE_PREFIX_PATH' (line 272)
Dockerfile.aarch64:155

 => ERROR [common 10/19] RUN python3 -m pip install -U     argcomp  63.8s
--------------------
 154 |     # Python3 PIP packages
 155 | >>> RUN python3 -m pip install -U \
 156 | >>>     argcomplete \
 157 | >>>     autopep8 \
 158 | >>>     flake8 \
 159 | >>>     flake8-blind-except \
 160 | >>>     flake8-builtins \
 161 | >>>     flake8-class-newline \
 162 | >>>     flake8-comprehensions \
 163 | >>>     flake8-deprecated \
 164 | >>>     flake8-docstrings \
 165 | >>>     flake8-import-order \
 166 | >>>     flake8-quotes \
 167 | >>>     gpustat==0.6.0 \
 168 | >>>     importlib_resources \
 169 | >>>     networkx \
 170 | >>>     "numpy>=1.24.4,<2" \
 171 | >>>     numpy-quaternion \
 172 | >>>     onnx \
 173 | >>>     pydocstyle \
 174 | >>>     pymongo \
 175 | >>>     pyyaml \
 176 | >>>     "scipy>=1.7.0" \
 177 | >>>     scikit-image \
 178 | >>>     scikit-learn \
 179 | >>>     "setuptools_scm>=6.2" \
 180 | >>>     tqdm \
 181 | >>>     trimesh \
 182 | >>>     "warp-lang>=0.9.0" \
 183 | >>>     "yourdfpy>=0.0.53" \
 184 | >>>     opentelemetry-api==1.28.1 \
 185 | >>>     opentelemetry-sdk==1.28.1 \
 186 | >>>     opentelemetry-exporter-otlp
 187 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c python3 -m pip install -U     argcomplete     autopep8     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     gpustat==0.6.0     importlib_resources     networkx     \"numpy>=1.24.4,<2\"     numpy-quaternion     onnx     pydocstyle     pymongo     pyyaml     \"scipy>=1.7.0\"     scikit-image     scikit-learn     \"setuptools_scm>=6.2\"     tqdm     trimesh     \"warp-lang>=0.9.0\"     \"yourdfpy>=0.0.53\"     opentelemetry-api==1.28.1     opentelemetry-sdk==1.28.1     opentelemetry-exporter-otlp" did not complete successfully: exit code: 1
~/workspaces/isaac_ros-dev/src/isaac_ros_common

ArenaGrenade, 2025-04-05:
I got a different error at the same step. Something related to packaging and none of the common fixes help.

```
#17 35.70 Collecting mapbox_earcut>=1.0.2
#17 35.72   Downloading mapbox_earcut-1.0.3.tar.gz (24 kB)
#17 35.98   Installing build dependencies: started
#17 39.95   Installing build dependencies: finished with status 'done'
#17 39.96   Getting requirements to build wheel: started
#17 40.20   Getting requirements to build wheel: finished with status 'done'
#17 40.20   Preparing metadata (pyproject.toml): started
#17 40.48   Preparing metadata (pyproject.toml): finished with status 'error'
#17 40.50   error: subprocess-exited-with-error
#17 40.50   
#17 40.50   × Preparing metadata (pyproject.toml) did not run successfully.
#17 40.50   │ exit code: 1
#17 40.50   ╰─> [30 lines of output]
#17 40.50       Traceback (most recent call last):
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 515, in validate
#17 40.50           packaging.utils.canonicalize_name(self.name, validate=True)
#17 40.50       TypeError: canonicalize_name() got an unexpected keyword argument 'validate'
#17 40.50       
#17 40.50       During handling of the above exception, another exception occurred:
#17 40.50       
#17 40.50       Traceback (most recent call last):
#17 40.50         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
#17 40.50           main()
#17 40.50         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
#17 40.50           json_out['return_val'] = hook(**hook_input['kwargs'])
#17 40.50         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 164, in prepare_metadata_for_build_wheel
#17 40.50           return hook(metadata_directory, config_settings)
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/__init__.py", line 96, in prepare_metadata_for_build_wheel
#17 40.50           return _build_wheel_impl(
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/wheel.py", line 174, in _build_wheel_impl
#17 40.50           return _build_wheel_impl_impl(
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/wheel.py", line 231, in _build_wheel_impl_impl
#17 40.50           metadata = get_standard_metadata(pyproject, settings)
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/metadata.py", line 75, in get_standard_metadata
#17 40.50           metadata = StandardMetadata.from_pyproject(
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 428, in from_pyproject
#17 40.50           self = cls(
#17 40.50         File "<string>", line 24, in __init__
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 293, in __post_init__
#17 40.50           self.validate()
#17 40.50         File "/tmp/pip-build-env-g77x57aj/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 516, in validate
#17 40.50           except packaging.utils.InvalidName:
#17 40.50       AttributeError: module 'packaging.utils' has no attribute 'InvalidName'
#17 40.50       [end of output]
#17 40.50   
#17 40.50   note: This error originates from a subprocess, and is likely not a problem with pip.
#17 40.50 error: metadata-generation-failed
#17 40.50 
#17 40.50 × Encountered error while generating package metadata.
#17 40.50 ╰─> See above for output.
#17 40.50 
#17 40.50 note: This is an issue with the package mentioned above, not pip.
```

Sean-Marriott, 2025-04-07:
I am having the same issue as @ArenaGrenade on two separate Jetson AGX Orins.

kevinchan04, 2025-04-07:
I am also having this issue. This issue relates with pip install of `yourdfpy`. Delete it from the dockerfile may help.

padmakumar76, 2025-04-08:
while running the code on Jetson Orin Nano, getting the same error explained by arthurgritzky  and ArenaGrenade

joewong00, 2025-04-08:
Getting the same error. Any fixes??

IshaanSaigal, 2025-04-08:
I am also getting the same error on a new Orin NX.

arthurgritzky, 2025-04-08:
> I am also having this issue. This issue relates with pip install of `yourdfpy`. Delete it from the dockerfile may help.

Thanks a lot it works now.

arthurgritzky, 2025-04-08:
The fix that worked for me was to comment out line 183 in the Dockerfile (the line that says: "yourdfpy>=0.0.53" \ ).
(as kindly pointed out by **@kevinchan04** )

Depending on your platform in:

 **Dockerfile.aarch64** (for Jetson)

 **Dockerfile.x86_64** (for x86)

These files are located in:
cd ${ISAAC_ROS_WS}/src/isaac_ros_common/docker`

Just open the appropriate Dockerfile with a text editor, navigate to line **183** ("yourdfpy>=0.0.53" \) and either remove or comment it out.

Hope this helps.

closed 2025-04-08 by arthurgritzky