NVIDIA-ISAAC-ROS/isaac_ros_common #187: Failure when running run_dev.sh on Jetson Orin Nano

pcarff, 2025-04-24:
I have walked through the output of the installation when running run_dev.sh and it appears that two lines are not executed:
packaging==24.2 \ 
"mapbox_earcut==1.0.1" \

As a result the install is failing.

pcarff@ubuntu:~/workspaces/isaac_ros_dev/src/isaac_ros_common$ ./scripts/run_dev.sh 
Launching Isaac ROS Dev container with image key aarch64.ros2_humble.realsense: /home/pcarff/workspaces/isaac_ros_dev
Building aarch64.ros2_humble.realsense base as image: isaac_ros_dev-aarch64
Building layered image for key aarch64.ros2_humble.realsense as isaac_ros_dev-aarch64
Using configured docker search paths: /home/pcarff/workspaces/isaac_ros_dev/src/isaac_ros_common/scripts/../docker
Checking if base image nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble-realsense_38f78b165b33c467e21fdf455fdd2165 exists on remote registry
Checking if base image nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_be39dbe798e68ecfcde559a2022a4b60 exists on remote registry
Checking if base image nvcr.io/nvidia/isaac/ros:aarch64_7b30ce7fcacb04cc3cd9b787c037d5fc exists on remote registry
Resolved the following 3 Dockerfiles for target image: aarch64.ros2_humble.realsense
/home/pcarff/workspaces/isaac_ros_dev/src/isaac_ros_common/scripts/../docker/Dockerfile.realsense
/home/pcarff/workspaces/isaac_ros_dev/src/isaac_ros_common/scripts/../docker/Dockerfile.ros2_humble
/home/pcarff/workspaces/isaac_ros_dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64
Building /home/pcarff/workspaces/isaac_ros_dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64 as image: aarch64-image with base: 
[+] Building 57.8s (15/48)                                                                docker:default
 => [internal] load build definition from Dockerfile.aarch64                                        0.0s
 => => transferring dockerfile: 17.74kB                                                             0.0s
 => [internal] load metadata for nvcr.io/nvidia/12.6.11-devel:12.6.11-devel-aarch64-ubuntu22.04     0.8s
 => [internal] load .dockerignore                                                                   0.0s
 => => transferring context: 2B                                                                     0.0s
 => [internal] load build context                                                                   0.0s
 => => transferring context: 179B                                                                   0.0s
 => [base-arm64 1/1] FROM nvcr.io/nvidia/12.6.11-devel:12.6.11-devel-aarch64-ubuntu22.04@sha256:eb  0.0s
 => CACHED [common  1/19] RUN mkdir -p /opt/nvidia/isaac_ros_dev_base && dpkg-query -W | sort > /o  0.0s
 => CACHED [common  2/19] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt-g  0.0s
 => CACHED [common  3/19] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt-g  0.0s
 => CACHED [common  4/19] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt-g  0.0s
 => CACHED [common  5/19] RUN update-alternatives --install /usr/bin/python python /usr/bin/python  0.0s
 => CACHED [common  6/19] RUN --mount=type=cache,target=/var/cache/apt     wget -qO - https://isaa  0.0s
 => CACHED [common  7/19] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt-g  0.0s
 => CACHED [common  8/19] RUN python3 -m pip install -U     setuptools==65.7.0                      0.0s
 => CACHED [common  9/19] RUN python3 -m pip install -U     Cython     ninja     wheel              0.0s
 => ERROR [common 10/19] RUN python3 -m pip install -U     argcomplete     autopep8     flake8     56.9s
------                                                                                                   
 > [common 10/19] RUN python3 -m pip install -U     argcomplete     autopep8     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     gpustat==0.6.0     importlib_resources     networkx     "numpy>=1.24.4,<2"     numpy-quaternion     onnx     pydocstyle     pymongo     pyyaml     "scipy>=1.7.0"     scikit-image     scikit-learn     "setuptools_scm>=6.2"     tqdm     trimesh     "warp-lang>=0.9.0"     "yourdfpy>=0.0.53"     opentelemetry-api==1.28.1     opentelemetry-sdk==1.28.1     opentelemetry-exporter-otlp:
1.611 Collecting argcomplete
1.778   Downloading argcomplete-3.6.2-py3-none-any.whl (43 kB)
1.817      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.7/43.7 KB 1.1 MB/s eta 0:00:00
1.919 Collecting autopep8
2.005   Downloading autopep8-2.3.2-py2.py3-none-any.whl (45 kB)
2.041      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.8/45.8 KB 1.2 MB/s eta 0:00:00
2.047 Requirement already satisfied: flake8 in /usr/lib/python3/dist-packages (4.0.1)
2.169 Collecting flake8
2.193   Downloading flake8-7.2.0-py2.py3-none-any.whl (57 kB)
2.208      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.8/57.8 KB 5.0 MB/s eta 0:00:00
2.269 Collecting flake8-blind-except
2.297   Downloading flake8-blind-except-0.2.1.tar.gz (3.7 kB)
2.317   Preparing metadata (setup.py): started
2.895   Preparing metadata (setup.py): finished with status 'done'
2.993 Collecting flake8-builtins
3.016   Downloading flake8_builtins-2.5.0-py3-none-any.whl (11 kB)
3.084 Collecting flake8-class-newline
3.109   Downloading flake8_class_newline-1.6.0-py3-none-any.whl (5.2 kB)
3.221 Collecting flake8-comprehensions
3.302   Downloading flake8_comprehensions-3.16.0-py3-none-any.whl (8.2 kB)
3.391 Collecting flake8-deprecated
3.415   Downloading flake8_deprecated-2.2.1-py3-none-any.whl (17 kB)
3.488 Collecting flake8-docstrings
3.512   Downloading flake8_docstrings-1.7.0-py2.py3-none-any.whl (5.0 kB)
3.607 Collecting flake8-import-order
3.629   Downloading flake8_import_order-0.18.2-py2.py3-none-any.whl (15 kB)
3.728 Collecting flake8-quotes
3.754   Downloading flake8-quotes-3.4.0.tar.gz (14 kB)
3.796   Preparing metadata (setup.py): started
4.334   Preparing metadata (setup.py): finished with status 'done'
[... 297 lines cut ...]
 166 | >>>     flake8-quotes \
 167 | >>>     gpustat==0.6.0 \
 168 | >>>     importlib_resources \
 169 | >>>     networkx \
 170 | >>>     "numpy>=1.24.4,<2" \
 171 | >>>     numpy-quaternion \
 172 | >>>     onnx \
 173 | >>>     pydocstyle \
 174 | >>>     pymongo \
 175 | >>>     pyyaml \
 176 | >>>     "scipy>=1.7.0" \
 177 | >>>     scikit-image \
 178 | >>>     scikit-learn \
 179 | >>>     "setuptools_scm>=6.2" \
 180 | >>>     tqdm \
 181 | >>>     trimesh \
 182 | >>>     "warp-lang>=0.9.0" \
 183 | >>>     "yourdfpy>=0.0.53" \
 184 | >>>     opentelemetry-api==1.28.1 \
 185 | >>>     opentelemetry-sdk==1.28.1 \
 186 | >>>     opentelemetry-exporter-otlp
 187 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c python3 -m pip install -U     argcomplete     autopep8     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     gpustat==0.6.0     importlib_resources     networkx     \"numpy>=1.24.4,<2\"     numpy-quaternion     onnx     pydocstyle     pymongo     pyyaml     \"scipy>=1.7.0\"     scikit-image     scikit-learn     \"setuptools_scm>=6.2\"     tqdm     trimesh     \"warp-lang>=0.9.0\"     \"yourdfpy>=0.0.53\"     opentelemetry-api==1.28.1     opentelemetry-sdk==1.28.1     opentelemetry-exporter-otlp" did not complete successfully: exit code: 1
~/workspaces/isaac_ros_dev/src/isaac_ros_common

AnjaKang, 2025-04-25:
I am also encountering with same issue, I am using orin nx

AnjaKang, 2025-04-25:
Hi, I have referred to NVIDIA Community [this](https://forums.developer.nvidia.com/t/nvblox-isaac-ros-setup-on-ros2-with-a-zed-mini-camera/329443) to change the Dockerfile.aarch64 and I passed.

pcarff, 2025-04-25:
@AnjaKang Thank you - it is still running and it is way further along than before!!

closed 2025-04-25 by pcarff