NVIDIA-ISAAC-ROS/isaac_ros_common #188: run_dev.sh failed.

Jouybari7, 2025-05-03:
run_dev.sh failed with the following error, 

I changes packaging==24.2 \ to packaging==23.2 \ in Dockerfile.aarch64 file and worked. I dont know if this is right though. 

ERROR: failed to solve: process "/bin/bash -c python3 -m pip install     argcomplete     autopep8     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     gpustat==0.6.0     importlib_resources     networkx     \"numpy>=1.24.4,<2\"     numpy-quaternion     onnx     pydocstyle     pymongo     pyyaml     \"scipy>=1.7.0\"     scikit-image     scikit-learn     \"setuptools_scm>=6.2\"     tqdm     trimesh     \"warp-lang==0.11.0\"     packaging==24.2     \"mapbox_earcut==1.0.1\"     \"yourdfpy==0.0.53\"     opentelemetry-api==1.28.1     opentelemetry-sdk==1.28.1     opentelemetry-exporter-otlp==1.28.1" did not complete successfully: exit code: 1
~/workspaces/isaac_ros-dev/src/isaac_ros_common

alexisurvoy, 2025-05-07:
Also ran into the same issue - changing "RUN python3 -m pip install -U \" to "RUN python3 -m pip install --use-deprecated=legacy-resolver -U \" solved it for me.

naslevente, 2025-05-09:
I can confirm the solution presented by @alexisurvoy worked for me as well. My concrete error was the following:

```
Collecting xatlas
135.5   Downloading xatlas-0.0.10.tar.gz (7.7 MB)
137.0      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.7/7.7 MB 5.0 MB/s eta 0:00:00
138.3   Installing build dependencies: started
142.9   Installing build dependencies: finished with status 'done'
142.9   Getting requirements to build wheel: started
143.2   Getting requirements to build wheel: finished with status 'done'
143.2   Preparing metadata (pyproject.toml): started
143.6   Preparing metadata (pyproject.toml): finished with status 'error'
143.7   error: subprocess-exited-with-error
143.7   
143.7   × Preparing metadata (pyproject.toml) did not run successfully.
143.7   │ exit code: 1
143.7   ╰─> [30 lines of output]
143.7       Traceback (most recent call last):
143.7         File "/tmp/pip-build-env-4wrua6lr/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 515, in validate
143.7           packaging.utils.canonicalize_name(self.name, validate=True)
143.7       TypeError: canonicalize_name() got an unexpected keyword argument 'validate'
```

Simply added the `--use-deprecated=legacy-resolver` to the `pip install` command in the `Docker.aarch64` file.

jaiveersinghNV, 2025-05-20:
Thanks for reporting this issue. We believe this was fixed with our Update 8 released last week. Could you please pull the latest from `main` and try again?
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/releases/tag/v3.2-8